pybricks/support #1560: Potential bug with use of comment blocks [Bug]

MonongahelaCryptidCooperative, 2024-03-22:
[image]
So I found out that if you put a comment block in an empty conditional (such as a elif of else: the default value of pass will not be placed. This obviously results in a syntax error. I have included a screenshot showing this (there is a comment but no "pass" created with from the block code. 

Not a huge deal and I figured out what was going on immediately but could be an issue for novice users.

laurensvalk, 2024-03-22:
Total rookie mistake on my part! Thanks for catching this. Will be fixed soon.

laurensvalk, 2024-03-22:
Let's see if we've now covered all cases:

[image]

[image]

[image]

[image]

[image]

-----------------

This is still broken:

[image]

laurensvalk, 2024-03-22:
So a comment block (like any block) suppresses the normal generation of the `pass` keyword.

Instead of messing with that in every possible parent block, we can just add the `pass` back below the comment if:
- the comment block is the last in a stack &&
- there are no other types of block in the stack &&
- the comment isn't in a loop/function where we already add a loop trap (`wait(1)`) in multitask mode &&
- it isn't the main block in a synchronous program

closed 2024-03-22 by laurensvalk

laurensvalk, 2024-03-22:
Fix released in 

[image]