pybricks/support #1561: pybricks beta APP does not update where the website on chrome does

BertLindeman, 2024-03-22:
**Describe the bug**
The installed APP stays at V2.4.0-beta-7 even after stopping the APP 

**To reproduce**
Steps to reproduce the behavior:
1. Start the (from chrome) installed APP (on Win11)
2. Click on Check for update 
3. See a rotating circle and wait for it to finish
4. Click on 'about'
5. See the version V2.4.0-beta7

**Expected behavior**
APP updated like on the website
Did the scenario before and after update of the website.
Same result

**Screenshots**
chrome updated on top, APP below:
[image]

laurensvalk, 2024-03-22:
Is the app using the same browser? Or perhaps it is using Edge, or a different Chrome browser profile?

Usually, you can just close all browser windows and the next time you get the new version.

I stopped making early announcements about new releases so everyone gets there update gradually :wink: (Except for the enthusiasts following us hear :smile: )

BertLindeman, 2024-03-22:
Would be handy if I could see which browser the app is using.
Ah the shortcut on the desktop:
`"C:\Program Files\Google\Chrome\Application\chrome_proxy.exe"  --profile-directory=Default --app-id=` 
removed the app-id as not interesting.

Looked in Edge, got immed at `v3.5.0b1 (Pybricks Beta v2.5.0-beta.1)` and in Edge there is the "install as app"
So assume the app is not using edge.
Same on Opera
Same on Firefox (this got me to go to the welcome tour, was never used in firefox)

Stopped all browsers except firefox where I type this....

Just started the app still old version
Did "back to stable" and "try beta" still old version.

Next I would uninstall / reinstall the app. But how?
Ok, from the 3-dots menu do `uninstall pybricks beta`
[image]
Did not remove from Chrome.

Back in Chrome on beta.pybricks.com:
`install as app` 
`pin to taskbar`

And success (sort off - would want to know WHY):
[image]

For me this issue can be closed.
Keep this open for further research?

laurensvalk, 2024-03-22:
I think we've established that it does usually come eventually, it can just take some time.

The next time, keep an eye on the network tab of the dev tools in the browser:

[image]

When there is a new version and do 'check for update', you'll see a lot of activity there while it spins.

closed 2024-03-22 by laurensvalk

laurensvalk, 2024-03-22:
@BertLindeman  - a new version is already here, so there's your chance :smile: 

It's a small update (no new docs) so this one should load pretty quickly.

BertLindeman, 2024-03-22:
> @BertLindeman - a new version is already here, so there's your chance 😄
> 
> It's a small update (no new docs) so this one should load pretty quickly.

First stopped the chrome browser and then "check for updates"

[image]

restart..
[image]