pybricks/support #1571: [Bug] Paste blocks near cursor

laurensvalk, 2024-03-28:
> Currently pasted blocks appear close to the position of the copied original. This might even lead to the situation that the pasted blocks appear outside of the view port and the user thinks that it was not pasted at all.

_Originally posted by @ggramlich in https://github.com/pybricks/support/discussions/1568_

laurensvalk, 2024-03-28:
@ggramlich: Feel free to create these as issues directly. Basically, anything that has a "yes we can" or "no we can't" answer can just be an issue, so don't hold back :smile:  Everything else ("How/why...") can be a discussion.

laurensvalk, 2024-03-29:
Good suggestion, though a good bit of work will be needed to make this work. Ideally, we'd use something built into Blockly for this.

I think this could work well for copy paste via the right-click context menu. On pasting, I suppose we could move the stack to the location of the context menu.

But what about ctrl+c, ctrl+v? @ggramlich, what would you intuitively expect to happen in that case? Should it be pasted where the workspace was clicked last?

laurensvalk, 2024-03-29:
It looks like some work has been done upstream in https://github.com/google/blockly/pull/7521

It doesn't seem to be merged into a release yet.

BertLindeman, 2024-03-29:
A scenario I might do:
1. select a block
2. ctrl-c
3. move to another part of the program
4. crtl-v

and 3 and 4 repeated a few times to paste the copied part(s) to different places.

As example a part of debugging or such.
Would be hard to find out at what location I want it to go.
Most of the times I just paste a few times and move it where I want it to go.

ggramlich, 2024-04-07:
> It looks like some work has been done upstream in [google/blockly#7521](https://github.com/google/blockly/pull/7521)
> 
> It doesn't seem to be merged into a release yet.

The behavior there seems to be exactly as I would expect it. It looks like your comment there caused some movement.

ggramlich, 2024-05-24:
@laurensvalk 

The blockly issue is closed now.
https://github.com/google/blockly/pull/8173
https://github.com/google/blockly/issues/6848

laurensvalk, 2024-05-24:
Thanks for the heads up! I guess we can pull it in via https://github.com/pybricks/support/issues/1572