pybricks/support #1577: [Bug] Failed to execute 'showSaveFilePicker' on 'Window'

BertLindeman, 2024-04-01:
**Describe the bug**
In code.pybricks.com I want to save all sources in a zip, so press the `Backup all files`

Results in 
```
Error: Failed to execute 'showSaveFilePicker' on 'Window': Must be handling a user gesture to show a file picker.
    at Object.default (https://code.pybricks.com/static/js/main.af0af6bc.js:2:282930)
    at Ol (https://code.pybricks.com/static/js/main.af0af6bc.js:2:279640)
```

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com (on my win11 23H2 system)
2. Click on 'Backup all files'
3. See the error as above.

**Expected behavior**
What did you expect to happen instead?
See a file/directory selection panel to safe the zip file.

**Screenshots**
None sofar

**Extra info**

About Pybricks Code:
`v3.4.0 (Pybricks Code v2.4.0)`

I just did the same on beta.pybricks.com with no problem.
But the files there are of course different.

I did this both on code and beta in Google Chrome Version 123.0.6312.86 (Official Build) (64-bit)

No indication of an error in developers console.
But 1 issue shown, no idea if it has any importance.
Screenshot:
[image]

Might this be a time-out due to too many files?

BertLindeman, 2024-04-01:
After starting `chrome://device-log` in which I did not see anything related (no idea what to look for).
Tried another time. Now the backup works.
Hard to test now I think....

Although......

It takes time before I as user see anything happening after clicking the backup button.

Now tried to click the button twice (ample time for that) and the here reported error pops-up again.

So user-error induced by late feedback to the button click.

Similar to click on the run button on a larger program and clicking again
Results in ````
GATT Error Unknown.
````

Issue can be closed, but maybe there could be some feedback on the button click?

dlech, 2024-04-01:
> Might this be a time-out due to too many files?

I don't think so. This is likely a race condition that we've been lucky to avoid so far. We use the Redux Sagas library to handle events in Pybricks Code. If there are no queued events, things are usually handled synchronously. But occasionally, something can be blocking the queue and handling an action can get deferred. Certain actions, like the file save dialog and the Blueooth dialog require direct user interaction to trigger them for security reasons. So when these get deferred, they don't work as you have seen.

I'm not really sure what we can do at this point without some major re-architecting of Pybricks Code. 

But fortunately, the problem seems to happen quite rarely.