pybricks/support #1584: [Bug]

BobCohenRobert, 2024-04-04:
**Describe the bug**
My Lego Technic hub cannot connect to Bluethooth using https://code.pybricks.com/
Getting error "Permission denied"
When trying to connect directly from my Windows 10 setting it connected for 2-3 seconds and then disconnect.
See attached error.
[image]
 
Sincerely,
Reuven Cohen
bob.cohen77@gmail.com

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

dlech, 2024-04-04:
What web browser and version are you using?

> When trying to connect directly from my Windows 10 setting it connected for 2-3 seconds and then disconnect.

Yeah, don't do this. Only connect directly from the web browser. Otherwise it messes things up. If the hub is still added in the Windows settings, remove it.

> Getting error "Permission denied"

Also see https://github.com/orgs/pybricks/discussions/270 for more troubleshooting tips.

I don't think we've seen this error before, so if removing it from your Windows settings doesn't fix it, we will probably need Bluetooth packet logs to troubleshoot it any further.

laurensvalk, 2024-09-16:
We'll close this issue due to inactivity. Please feel free to re-open if this still occurs.

closed 2024-09-16 by laurensvalk