pybricks/support #1585: [Bug]  Cityhub and movehub get stuck running build 3360

BertLindeman, 2024-04-04:
**Describe the bug**
On build 3360 testing remote disconnect hub gets stuck. Need to take battery out to stop
 the hub. Not sure yet where the problem is, so kept the description vague.

**To reproduce**
Steps to reproduce the behavior:
1. Install CI build 3360 on a Cityhub and / or MoveHub
2. Start a program like the one below this scenario
3. wait a few minutes - not sure if I need to connect a remote or not.
4. Not SURE how to really be sure the hub gets stuck
5. try to shutdown the hub by holding the hub button
6. see rapid blue flashing
7. take the batteries out

**Expected behavior**
What did you expect to happen instead?
The hub would either shutdown on power-low or just run on as there is no-limit remote time-out specified 

<details><summary>Test program for remote-disconnect on a cityhub / movehub</summary>

```python

from pybricks.hubs import ThisHub
from pybricks.pupdevices import Remote, Motor
from pybricks.parameters import Button, Color, Port
from pybricks.tools import wait

print('\nProgram movehub_remote_disconnect')

# Replace this with your hub. <----
hub = ThisHub()

# Replace this with a unique color <----
pair_color = Color.GREEN

while True:

    # Blink to indicate we want to connect.
    hub.light.blink(pair_color, (100, 200))

    # Try to connect, and retry on timeout.
    remote = Remote(timeout=None)

    # Light solid to indicate we are connected.
    remote.light.on(pair_color)
    hub.light.on(pair_color)
    col_index = 0
    colors = [Color.BROWN, Color.MAGENTA, Color.RED, Color.GRAY, Color.BLUE]

    while True:
        # Wait until a button is pressed.
        pressed = remote.buttons.pressed()

        if Button.CENTER in pressed:
            break
        elif pressed:#
            cur_color = colors[col_index]
            remote.light.on(cur_color)
            hub.light.on(cur_color)
            col_index += 1
            if col_index >= len(colors):
                col_index = 0
            while pressed:  # wait until no button pressed
                pressed = remote.buttons.pressed()
                wait(10)

        # else:
        #     pass

        # Wait some time.
        wait(100)

    # Disconnect.
    hub.light.on(Color.YELLOW)
    remote.disconnect()
    print("disconnected")

    # Give other hub a chance to connect.
    wait(2000)
    print('Wait time passed')

```

</details>

**Screenshots**
None now.

BertLindeman, 2024-04-05:
Could not get a scenario for this one, as I do not understand what the cause might be.

First hit was on the syntax error where I had the old ```remote.close()``` instead of ```remote.disconnect()```

But there must be something else needed. 
Loss of the remote does not trigger this (e.g. remote out of range) program stops for missing device.

Yet I had this on both the cityhub and the movehub.

Might this be related to:
```remote = Remote(timeout=None)```
The comment there states:
    # Try to connect, and retry on timeout.
In the firmware is there a retry from time to time? How often is that?

Will retry

laurensvalk, 2024-04-05:
Which color was the hub light before it started flashing when you shut it down?

BertLindeman, 2024-04-05:
I think it was green blinking, as I thought it was waiting for a remote connection.
Then tried to stop the program via the ui. 
No reaction, so hold-hub-button.

I am not sure. And did not note it down.

laurensvalk, 2024-04-05:
That does seem the most likely. We didn't change the connect step, though, so this may be an artifact of multiple hubs trying to connect to the same remote at the same time in this specific scenario (made possible by the disconnect method).

BertLindeman, 2024-04-05:
Frustrating that the first test on both hubs fails and I cannot reproduce. Grr.
I was swapping remote batteries in that time, so they might have been trying to connect quite some time.
There was in that time frame only one hub active at each time.

laurensvalk, 2024-04-05:
I was able to reproduce the lockup with this program on both hubs and semi-aggressively restarting the programs with the buttons on the hub.

```python
from pybricks.hubs import ThisHub
from pybricks.pupdevices import Remote
from pybricks.tools import wait
from pybricks.parameters import Color

hub = ThisHub()

remote = Remote(timeout=None)
hub.light.on(Color.RED)
wait(5000)
```

So we might have had this one for a while.

BertLindeman, 2024-04-05:
Wow. 
How do you know it locked up?
Do you connect the remote on this test?

laurensvalk, 2024-04-05:
> Wow. How do you know it locked up? Do you connect the remote on this test?

I left the remote on, and kept pressing the buttons on the hub. Sometimes one hub would win, sometimes the other. Eventually the city hub program would not be turned off anymore.

laurensvalk, 2024-04-05:
I'm not too worried about this particular contrived scenario with pressing buttons aggressively but I'd like to further improve our ability to properly cancel programs.

Yesterday's updates ensured the final cleanups can't hold it up, but this must still be happening somewhere in the program, presumably in the scan+connect phase, which has a few non cancellable parts.

But those should still respect system shutdown, but aren't :smile:

BertLindeman, 2024-04-05:
I got the movehub **slowly blinking blue** and it was locked-up.

In your test program the remote light gets blue. Is that usual?
I thought it would stay white after connect. 

> But those should still respect system shutdown, but aren't 😄

It's horrible if you have a model that is not designed well enough for battery replacements.

laurensvalk, 2024-04-05:
> In your test program the remote light gets blue. Is that usual? I thought it would stay white after connect.

Blue is the default. Always has been :smile: 

> It's horrible if you have a model that is not designed well enough for battery replacements.

I'm on it! :smile:

laurensvalk, 2024-04-05:
Fortunately, it's very easy to get something reproducible by adding an artificial non-cancellable wait:

```diff
diff --git a/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c b/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
index 3f1042659..679af3f89 100644
--- a/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
+++ b/lib/pbio/drv/bluetooth/bluetooth_stm32_cc2640.c
@@ -499,6 +499,7 @@ static PT_THREAD(peripheral_scan_and_connect_task(struct pt *pt, pbio_task_t *ta
     assert(peri->notification_handler);
 
     static pbio_error_t connection_error;
+    static struct etimer hack_timer;
 
     PT_BEGIN(pt);
 
@@ -515,6 +516,11 @@ static PT_THREAD(peripheral_scan_and_connect_task(struct pt *pt, pbio_task_t *ta
         }
     }
 
+    PROCESS_CONTEXT_BEGIN(&pbdrv_bluetooth_spi_process);
+    etimer_set(&hack_timer, 10000);
+    PROCESS_CONTEXT_END(&pbdrv_bluetooth_spi_process);
+    PT_WAIT_UNTIL(pt, etimer_expired(&hack_timer));
+
     // Optionally, disconnect from host (usually Pybricks Code).
     if (conn_handle != NO_CONNECTION &&
         (peri->options & PBDRV_BLUETOOTH_PERIPHERAL_OPTIONS_DISCONNECT_HOST)) {
~
```

This mimics any non-cancellable wait in the Bluetooth tasks. (Which is probably what we induced by two hubs connecting to one remote at the same time).

While we can't make all of them cancellable, I'd like to be able to exit these at least on shutdown.

BertLindeman, 2024-04-05:
Also hits on TecnicHubs.
Could not get the prime/inventor hub to hit. (My arm is getting tired :wink: )

Maybe I have some time to test this morning. Otherwise this evening.
Library duty is getting in the way.

BertLindeman, 2024-04-05:
### Confirm fix to work at build 3368.

### Symptoms: 
Hub seems to be running (fading slow blue) but does not react on button press.
Holding the hub button takes a few seconds and than a short fast blinking (blue) and OFF. 

TL;DR

### Scenario:
1. Run "issue_1585_laurens.py" on two (or more hubs)
2. Activate a remote and see one of the hubs led RED
3. Double press that hub button to stop / restart
4. see either hub taking the remote and turn red.
5. if the same hub gets "red" the other might be stuck
6. try double press on that hub to see if it reacts
7. hold the hub button to force shutdown
8. see that it does get OFF

<details><summary>Program  issue_1585_laurens.py</summary>

```python

from pybricks.hubs import ThisHub
from pybricks.pupdevices import Remote
from pybricks.tools import wait
from pybricks.parameters import Color
from pybricks import version

hub = ThisHub()
print("\nProgram issue_1585_laurens\n", version)
remote = Remote(timeout=None)
hub.light.on(Color.RED)
wait(5000)
```

</details>

Seen on CityHub, MoveHub and TechnicHub

```
 ('cityhub', '3.5.0b1', 'ci-build-3368-v3.5.0b1-15-g13dd65d6 on 2024-04-05')
 ('movehub', '3.5.0b1', 'ci-build-3368-v3.5.0b1-15-g13dd65d6 on 2024-04-05')
 ('technichub', '3.5.0b1', 'ci-build-3368-v3.5.0b1-15-g13dd65d6 on 2024-04-05')
```

laurensvalk, 2024-04-05:
:tada: !