pybricks/support #1588: [Bug] xbox cotroller disconnect with pybricks

WuHank, 2024-04-07:
Please do me a favor.

**Describe the bug**

xbox controller can connect with Microsoft surface go 3, but disconnect with the pybricks

**To reproduce**
1. have update the xbox controller 

2. the version of pybricks is:
MicroPython for LEGO® Powered Up smart hubs.
v3.4.0 (Pybricks Code v2.4.0)

3. type the basic code to ensure it as follow: 

from pybricks.iodevices import XboxController
from pybricks.parameters import Button

# Set up all devices.
xbox = XboxController()

# The main program starts here.
if Button.A in xbox.buttons.pressed():
    print('A')
elif Button.Y in xbox.buttons.pressed():
    print('Y')
else:
    pass

**Expected behavior**

Is the controller problem? 
or the model of the controller can't work?

**Screenshots**
1. xbox controller does conncet with the surface go 3:
[image]

2. xbox controller have updated:
[image]

3. The version of Pybricks:
[image]

4. The pic of pybricks and the program that tests the connection:
[image]

Thank you!

BertLindeman, 2024-04-07:
It will be very hard to see if your xbox controller does connect.

As soon as the xbox controller does connect, the program is ready in milliseconds.
There is no time to see anything.
 
I am not sure:
1. no connection - can be. did you press the connect button on the controller to allow it to connect to the spkie hub?
2. if connected - the program does not allow you time to do anything.

You might add a loop to do this test over and over.

Maybe like this:

[image]

Or even: 

[image]

WuHank, 2024-04-07:
> I am not sure:
> 
> 1. no connection - can be. did you press the connect button on the controller to allow it to connect to the spike hub?

I thought the controller only need to be connected with the surface go 3.
Could you tell me how to confirm the controller does connect with the spike prime?

> 2. if connected - the program does not allow you time to do anything.
> 
> You might add a loop to do this test over and over.

Thank for your help.  
You're so kind!
I also try the Pybricks Beta version.
But it still can't work.
Could you help me?
Thank you!

1. try the Pybricks Beta version
[image]

2. Thanks for your suggestion.  I fix the code but it still can't work.
[image]

Thanks a lot!
Please help me!

BertLindeman, 2024-04-07:
> I thought the controller only need to be connected with the surface go 3.
Could you tell me how to confirm the controller does connect with the spike prime?

The xbox controller does NOT connect to the PC or table or so, it connects to the hub.
In your case the spike hub.

From the pybricks documentation:
```
Xbox Controller Pairing Instructions
[¶](https://docs.pybricks.com/en/latest/iodevices/xboxcontroller.html#xbox-controller-pairing-instructions)

The first time you use a controller with a hub, you will need to pair them: 
Turn the controller on and then press and hold the pairing button on the back of the controller for a few seconds. When you release it, the Xbox button starts flashing more rapidly. 
Then start your program.

When pairing and the connection is succesful, the Xbox button will stop flashing and stay on for as long as the program is running.
```

To confirm the connection is there, you could add a block to change the color of the hub button,
or display some text on the hubs display.
Like this one:
[image]

So you would see immediately success.

WuHank, 2024-04-07:
> So you would see immediately success.

It's amazing!
It works!
Thank you so much!
Have a good day!

closed 2024-04-07 by WuHank

BertLindeman, 2024-04-07:
> > So you would see immediately success.
> 
> It's amazing! It works! Thank you so much! Have a good day!

You make my day! 😄 

It is hard to explain the various blinking of the controller.
Could you explain what made the difference for you?

WuHank, 2024-04-07:
> Could you explain what made the difference for you?

As my observation, there are 2 modes of blinking.
The rhythm are differents:
1. Before press the pairing button.
The rhythm is triplets.
1st and 2nd beat are flashing.

2. After press the pairing button.
The rhythm is like disco.
It's 4 beat.(also 2 beat, up beat and down beat)

After all, like the manual say.
When pairing and the connection is succesful, the Xbox button will stop flashing.

I'm so happy it finally done.
I almost give up.
Thank you !

laurensvalk, 2024-04-08:
Glad it worked @WuHank  and thank you @BertLindeman  for helping!