pybricks/support #1590: Block to Python Generator Multiplies Arguments in `multitask until`

DrTom, 2024-04-07:
The `multitask until` block creates a `subtask` function in the context of async code. When a single argument is reference inside, the `subtask` argument gets multiplied with the same name seemingly by the time the argument is  used inside.  

[image]

laurensvalk, 2024-04-07:
This is going to the top of things to fix for this week's release. Thanks!

laurensvalk, 2024-04-07:
Aside from that bug, your program has another interesting aspect if you remove the 1000 ms wait.

That leaves the task empty, so it could be ignored, at least in the wait for all multitask case. But not in the `one task` case since it still has to win the "race" against the task that takes a nonzero amount of time to run. I'll fix that as well.

laurensvalk, 2024-04-07:
Fixed in the latest beta release. Thanks again!

[image]

closed 2024-04-07 by laurensvalk