pybricks/support #1613: [Feature] Is there any function can directly contorl the PWM of each port?

a10036gt, 2024-04-27:
Is it possible to implement a "pwm" method to drive the pin 1 and 2?

Not like the motor.dc() command, I want to build a custom sensor but I want using the pwm power at P1 P2 at same time, but due to Auto ID Check function, I can't use motor.dc() command and PUPDevice command at same time, is there any solution?

Thanks!

a10036gt, 2024-04-27:
Yes I emulating the Powered Up protocol on arduino and STM8.

a10036gt, 2024-04-27:
That's because we want to using the PWM power (battery power) to driving some heavy load parts, like Pixy 2 Camera or something like that. Because 3v3 may not have that high current to push these devices.

a10036gt, 2024-04-27:
Like SPIKE Ultrasonic and Color sensor, their LEDs is actually powered by P1 P2 100% PWM.

dlech, 2024-04-27:
It isn't actually possible to turn on both outputs at the same time. This is the table from the motor driver datasheet.

[image]

The LEGO sensors/lights will bring either P1 or P2 high, but not both at the same time.

If you are using the LEGO UART protocol, then you can set the appropriate flag for the pin you want to use for power in the extended info that is embedded in `INFO_NAME`. See https://github.com/pybricks/technical-info/blob/master/uart-protocol.md#info_name

[image]

a10036gt, 2024-04-27:
Oh I didn't noticed that appropriate flag before, my bad, I will try to add that into my code, thank you so much!

closed 2024-05-01 by a10036gt