pybricks/support #1645: [Bug] Using lighting cord causes devices to not be detected

MrWonka, 2024-05-29:
**Describe the bug**
[Using this cable](https://www.lightmybricks.com/collections/diy-power-lego-lights/products/lego-powered-up-cable-power-functions-2-0), which steals some power from the PU connector to run lighting, if I use a motor with the original firmware it works fine, but using pybricks it throws this error:

--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

I've gotten the same results with both the city and spike prime hubs. I realize that this may be too obscure of an issue, but a fix would be appreciated.

dlech, 2024-05-29:
Did you try the `PUPDevice()` class to see what the `id` is?

https://docs.pybricks.com/en/latest/iodevices/pupdevice.html#pybricks.iodevices.PUPDevice

MrWonka, 2024-05-29:
Interesting. It sees it as a train motor, despite using a SPIKE Medium Angular Motor

dlech, 2024-05-29:
No, there isn't a way to force it. I'm guessing the cable is intentionally wired that way? Is the light wired to the same wires as the motor so that the light turns on only if the motor is running? If you don't care about speed control on the motor, you could use the `DCMotor` class since it thinks it is a train motor.

MrWonka, 2024-05-29:
Yes the light is directly tied to when the motor is running. I tested it using the DCMotor pupdevice and it works. Not having the motor know where it is messes up my build. Oh well.

MrWonka, 2024-05-29:
thanks for the help

closed 2024-05-29 by MrWonka