pybricks/support #1663: [Feature] Add a method to disconnect Xbox Controller

PoweredUP-AC, 2024-06-03:
**Is your feature request related to a problem? Please describe.**
I wanted to make the hub and the Xbox controller shut down whenever the Xbox logo button is pressed. I assumed that hub.system.shutdown() would lead to controller being turned off the same way it does when the program ends, but it stays in pairing mode instead. 

**Describe the solution you'd like**
As disconnecting a controller with a dedicated method might be useful in some scenarios, I think that adding something like xbox.disconnect() is the correct way to do it.(there must be a better name though) This function should disconnect the conroller from the hub and turn it off (described above). 
It seems to me that this function needs to be called on system.shutdown(), too.

laurensvalk, 2024-06-04:
Does the controller eventually turn off though?

As an alternative, you could perhaps end the program (`raise SystemExit`) instead of shutting the hub down. This should lead to a normal disconnect and the hub will automatically power off in a few minutes.

> I assumed that hub.system.shutdown() would lead to controller being turned off the same way it does when the program ends

I would have assumed that too.

laurensvalk, 2026-10-08:
This is now available. Thanks!

closed 2026-10-08 by laurensvalk