pybricks/support #1664: Technic hub wont run the program right after flash[Bug]

eadunes, 2024-06-05:
**Describe the bug**
when I have flashed the code to the hub and try to start the program it stays on blue blinking. wether I switch it of before or not.  but when I connect the hub the the pc/mac and start the code via website, then stop and disconnect, I can start the program from the hub. this only happens 1 time after flashing the code to the hub. tried also a 2nd hub. same behavior What is the problem?

**To reproduce**
see above

**Expected behavior**
The program should be able to run directly after flash.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

laurensvalk, 2024-06-05:
Can you share the program you are trying to run?

Perhaps you can start with a simple program that just turns the light on for a few seconds.

Then you'll know that saving works properly.

eadunes, 2024-06-05:
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Color, Port, Direction
from pybricks.tools import wait

# Initialize motor for loader
Loader = Motor(Port.A, Direction.CLOCKWISE)
Loader.run_until_stalled(-400, duty_limit=50)
Loader.reset_angle(0)

# Initialize and run motor for conveyor belt
Conveyor = Motor(Port.B, Direction.CLOCKWISE)
Conveyor.run(1250)

# Initialize the hub
hub = TechnicHub(observe_channels=[42])

while True:
    data = hub.ble.observe(42)

    if data is False:
        hub.light.on(Color.ORANGE)
        Loader.run_target(400, 0, wait=False)
    
    if data is True:
        hub.light.on(Color.GREEN)
        Loader.run_target(400, 210, wait=False)
        wait(4000)
    if data is None:
        hub.light.on(Color.RED)
    wait(100)

laurensvalk, 2024-06-05:
What did you mean by `flash`? Did you run it with the run (▶) button? Did you get any errors in the output window at the bottom when you ran it?

laurensvalk, 2024-06-05:
Ah, that's not necessary. After you run it with the play button, you can just start it again with the green button on the hub.

No need to re-flash. In fact, that removes your program.

I suppose you may be referring to the video from racing brick - two years ago the steps you described were still necessary, but we've made it a lot easier since then :smile:

eadunes, 2024-06-05:
no no, U misunderstood me.

I write code, store it to the hub. then connect the motors. it wont run the program unless I have run the code one time via the  website.

laurensvalk, 2024-06-05:
Can you try it with a simple program that just turns the light on?

For example, copy paste the example from here into a new program:

https://docs.pybricks.com/en/stable/hubs/technichub.html#turning-the-light-on-and-off

That should make it easy to see if saving it on the hub worked or not.

Debenben, 2024-06-05:
@eadunes If I understand you correctly, you would like to only flash your program to the hub and prevent it from running immediately afterwards.

The step you describe as flashing your program is probably only flashing pybricks "operating system", it no longer flashes any user-program. You only have to do this step once. Your program is then uploaded to the hub when you press the play button on the website and saved permanently when you shutdown the hub.

A solution might be to prevent your program from running properly by disconnecting the motor on port A or waiting for another button press at the start of your program. In case that is inconvenient in your build, you could use the pybricksdev command line tool for uploading. As far as I know you cannot only upload using the website (other than hacking its source code with a custom script).

eadunes, 2024-06-06:
> @eadunes If I understand you correctly, you would like to only flash your program to the hub and prevent it from running immediately afterwards.

No. I have to start it once via the website so that I can then start later from the hub. Before that, it cannot be started via the hub.

laurensvalk, 2024-06-06:
Have you tried the suggestion with the program that just blinks the light?

eadunes, 2024-06-06:
> Have you tried the suggestion with the program that just blinks the light?

not yet. will do it later.

eadunes, 2024-06-06:
> https://docs.pybricks.com/en/stable/hubs/technichub.html#turning-the-light-on-and-off

did run your suggested code. exact same behavior.

after saving the program to the hub its not possible to run the program by pressing the green button. 
switching the hub off and on again and try the run the program worked either.
connecting the hub to the pc and run the program works.
after disconnecting the hub from pc again its possible to run the program by pressing the green button of the hub.

tried it with two technic hubs.

laurensvalk, 2024-06-06:
Thanks for your response.

> after saving the program to the hub its not possible to run the program by pressing the green button

So running it worked and then starting again with the button did not work?

> after disconnecting the hub from pc again its possible to run the program by pressing the green button of the hub.

So running it worked and starting with the button did work this time? 

-----

I guess I still can't completely follow. Can you describe the difference between these two cases a above?

These are normally the same thing. Did it work once but not the second time?

Or did you mean it is only lost after you reboot the hub? Did you shut it down normally by holding the green button for about 3 seconds until it is off? 

I don't seem to be able to reproduce this issue on my Technic hub, so I'm trying to figure out what the difference is :slight_smile:

eadunes, 2024-06-08:
Hi

this isnt an issue anymore. it's me not knowing that saving the code to the hub procedure change and its not necessary anymore to bootload the hub to save the code inside firmware.

sorry for this.

Daniel

laurensvalk, 2024-06-08:
No worries - Glad it worked! 

I hope you'll like this easier method :smile:

laurensvalk, 2024-07-26:
Closed as resolved.

closed 2024-07-26 by laurensvalk