pybricks/support #1685: [Bug]

AlmazKad, 2024-06-23:
**Describe the bug**
Is City Hub incompatible with Xbox Controller?

**To reproduce**
1. Installed Pybricks Firmware on City Hub, used the code from the link:
https://pybricks.com/project/technic-42124-xbox/#running-it-as-a-python-program
2. tried to connect the Xbox Controller, and this happened:
Traceback (most recent call last):
  File "Buggy02.py", line 4, in <module>
ImportError: can't import name XboxController

Then repeated the same thing in the beta version, the result is the same error
But everything works as expected with other hubs

The question is, is City Hub compatible with Xbox Controller?

**Screenshots**
[image]

BertLindeman, 2024-06-23:
You are correct according to the current documentation: 

[image]

Tested with the current beta firmware:
`'cityhub', '3.5.0', 'v1.20.0-23-g6c633a8dd on 2024-04-11'`

AlmazKad, 2024-06-23:
Oh, thanks, didn't notice this page, my bad

laurensvalk, 2024-06-25:
Thanks @AlmazKad for asking and @BertLindeman  for answering! I'll close this issue.

closed 2024-06-25 by laurensvalk