pybricks/support #1690: On firmware install: Spike Prime hub did not show DFU device and Robot Inventor did

BertLindeman, 2024-06-26:
After some time, I wanted to flash newer firmware (for #1678) on the primehub that was still at **this** CI firmware (build 3380 I think).

1. Bluetooth is enabled on the hub.
2. Press the bluetooth button and insert the USB cable to Win11 and wait for a colorful led blinking.
3. No DFU device to connect to.
4. windows device manager shows an exclamation mark on the device.

Without any change on the windows side tried the same action on a Robot Inventor hub.
And **that** went fine.

Did a re-install of the dfu driver as you document.
Now the PrimeHub **is** shown as DFU device and can have it's firmware updated.

So no problem in the end, but I add this in case anyone has this problem too.

Bert

_Originally posted by @BertLindeman in https://github.com/pybricks/support/issues/1615#issuecomment-2190024249_

laurensvalk, 2024-06-26:
Doing the DFU driver steps seems to be required on Windows for every new hub you use.

I wonder if Windows "forgets" about certain devices eventually, so it would treat it like a new hub?

BertLindeman, 2024-06-26:
"eventually" is the key word I think.

The last hub I installed yesterday was the prime hub, so to test I now updated the RobotInventor Hub.
No problem.
So it's not really a simple one usb hub only.

dlech, 2024-06-26:
Robot Inventor and SPIKE Prime and SPIKE Essential all have different USB product IDs, so I would expect that we would have to configure the WinUSB driver for each type of hub individually.

Windows should not "forget" unless you manually uninstall the driver/remove the device from the *Device Manager*.

laurensvalk, 2024-06-26:
> I would expect that we would have to configure the WinUSB driver for each type of hub individually.

I think it is each hub, not just each hub type.

dlech, 2024-06-26:
Interesting, I wonder if Windows is looking at the serial number as well when we manually configure a driver.

laurensvalk, 2024-06-26:
Oh, nice - if that means that there might be a slightly different setup path that would make it necessary just once, that would be pretty neat for classroom use.

dlech, 2024-06-26:
For for that, I think https://github.com/pbatard/libwdi/wiki/Zadig can help.

laurensvalk, 2024-06-26:
I suppose we could test if that generic windows driver installer Zadig we had once upon a time makes it necessary once or for every hub.

dlech, 2024-06-26:
I wonder if we could use Zadig to create a `.inf` file and then just share that file rather than everyone having to install Zadig?

laurensvalk, 2024-06-26:
:rocket:

afarago, 2024-08-22:
> I wonder if we could use Zadig to create a `.inf` file and then just share that file rather than everyone having to install Zadig?

Tried it and does not work.

**Background**
Zadig creates a self-signed .cat catalog file and installs a custom root certificate for the local device.
When moving the file to another computer the root cert / trust provider is not there, thus this message will pop up.
[image]

**Per hub**
Unfortunately Windows does need this step every hub, therefore the text here is not entirely correct.
[image]
Last week I have done a workshop and flashed/re-flashed 8 devices, requiring a devmgr step 8 times.

laurensvalk, 2024-08-23:
Thank you for trying this Attila. To be complete, is Zadig required only once per computer, or also once per hub like the device manager method?