pybricks/support #1704: [Bug] Observe continues to pick up last broadcast even when nothing is sent

StPot3007, 2024-07-08:
**Describe the bug**
Using 2 hubs and one remote, remote1 & hub 1 send a broadcast to hub 2. Observe on hub 2 receives the broadcast from hub 1 and instructions get processed, but instead of then 'resetting' what is being received the observe instruction seems to continue to receive the last information even though nothing further is broadcast from hub1

**### Broadcast code:**
from pybricks.hubs import CityHub
from pybricks.pupdevices import DCMotor, Light, Remote,ColorDistanceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

# Color and brightness of Hub LEDs
LEDconn = Color.GREEN*0.3       # if Hub connected, color * brightness
LEDnotconn = Color.RED*0.5      # if Hub is not connect, color * brightness

hub = CityHub(2,observe_channels=[1])

hub.light.on(LEDnotconn)

print ('Hub name: ' + hub.system.name())

# -----------------------------------------------
# main loop
# -----------------------------------------------

remote_not_connected=True

try:
    # Connect to 'Decoupler' remote, wait max 10secs (default)
    my_remote = Remote("Decoupler")

    # Print the current name of the remote.
    print('Remote name: ' + my_remote.name())
    remote_not_connected=False
except OSError as ex:
    print('Shutdown')
    hub.system.shutdown()

hub.light.on(LEDconn)
my_remote.light.on(LEDconn)
remoteConnected = True
print('Remote Connected')

sensor = ColorDistanceSensor(Port.B)

while True:

    button = my_remote.buttons.pressed()
    pressed=str(button)
    print(pressed)

    if pressed!='set()':
        print('Pressed :' + pressed)
        hub.ble.broadcast(pressed)
        print("Broadcast button")
        my_remote.light.on(Color.YELLOW)
        my_remote.light.on(LEDconn)
    else:
        color=Color.NONE
        color=sensor.color()
        if color !=Color.NONE:
            print('Color :' + str(color))
            hub.ble.broadcast(str(color))
            print("Broadcast color")
        else:
            #hub.ble.broadcast(None)
            print('None')
    wait(100)
    #hub.ble.broadcast(None)

**### Observe code**
from pybricks.hubs import CityHub
from pybricks.pupdevices import DCMotor, Light, Remote
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

# Color and brightness of Hub LEDs
LEDconn = Color.GREEN*0.3       # if Hub connected, color * brightness
LEDnotconn = Color.RED*0.5      # if Hub is not connect, color * brightness

hub = CityHub(1,observe_channels=[1,2])

hub.light.on(LEDnotconn)

print ('Hub name: ' + hub.system.name())

# -----------------------------------------------
# main loop
# -----------------------------------------------

hub.light.on(LEDconn)

# Initialize the motor.
train_motor = DCMotor(Port.A)

while True:  #Loop indefinetly

    pressed=hub.ble.observe(2)

    if pressed is None:
        hub.light.on(Color.ORANGE)
    else:
        print(pressed)
        hub.light.on(Color.YELLOW)
        if pressed =="{Button.LEFT_PLUS}":
            train_motor.dc(50)
        if pressed == "{Button.CENTER}" or pressed == "Color.YELLOW":
            train_motor.stop()
            print('stop')

    #pressed=None
    wait(100)

**Expected behavior**
Once an observe instruction receives something it should reset until something new is received (so in the code above value of parameter 'pressed' should reset to 'None'

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

StPot3007, 2024-07-08:
> There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

Screenshot of printed values supposedly received by the 'observe' statement (the value of 'pressed')
[image]

laurensvalk, 2024-07-08:
> **Expected behavior**
> Once an observe instruction receives something it should reset until something new is received (so in the code above value of parameter 'pressed' should reset to 'None'

Broadcasting and observing don't really work like "sending an email", but more like repeatedly "speaking into an open phone line" in case somebody is listening. So the broadcaster keeps broadcasting the last value, and the observer keeps observing.

But you could add a few lines to wait for a value that isn't the same as before. I think this would achieve what you want?

StPot3007, 2024-07-08:
I see, this explains that 😁. I will do as you suggest then and check for changes in value rather than just picking up what is perceived by the observe statement. Thanks for the input.

closed 2024-10-04 by laurensvalk