pybricks/support #1705: [Bug] "Internal Error"

AW737, 2024-07-08:
Q: What is the problem?
A: When I open the page to the code (code.pybricks.com), it just shows an error message at the top, and 
after a second, the webpage just flashes to a white screen.

Steps to reproduce the behavior:
1. Go to 'code.pybricks.com'
2. See error

What did you expect to happen instead?
To open the page to let me code

There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
So, it's pretty hard to take a screenshot of it since I have a small window of taking it, so here's some of the page and the error message. After I see that, it just flashes to white, until I reload the page, which the same thing happens.

[image]

Don't know if I should put the error message here but, hey it's my first time.
```
UnknownError: Internal error.
 UnknownError: Internal error.
    at De.rejectionMapper (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1595134)
    at je (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1529120)
    at IDBOpenDBRequest.<anonymous> (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1556385)
    at IDBOpenDBRequest.<anonymous> (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1530914)
```

dlech, 2024-07-08:
What browser and version and operating system are you using?

dlech, 2024-07-08:
https://stackoverflow.com/questions/74207049/internal-error-opening-backing-store-for-indexeddb-open

suggests restarting the computer and making sure you aren't running the browser in "private" mode.

But basically, this is a browser issue. I don't think there is anything we can do to fix it on our end.

You could also check the console log in the browser for additional errors. That could help troubleshoot the problem better.