pybricks/support #1711: [Feature] Imporve number display for pybricks -499 to 499

afarago, 2024-07-14:
**Is your feature request related to a problem? Please describe.**
I am always a fan of showing numbers as part local feedback what the hub is doing.
SPIKE's display is limited to 5x5 pixels, and numeric display is suboptimal that I would love to see improved.

**Describe the solution you'd like**
I would like a better display for two digit numbers both positive and negative, a more straightforward one-digit display and solution to display numbers 0-360 range.

**Describe alternatives you've considered**
I have created a solution for displaying numbers between -499-499 that I would suggest to discuss and implement.

**Additional context**
python code: https://gist.github.com/afarago/90645439804b1075634186266a7bfdfd
detailed description and photos: https://digitalbrick.home.blog/2024/07/14/hot-summer-large-numbers-on-your-spike
video demo: https://youtube.com/shorts/ornyaFLwexo
[image]
[image]

laurensvalk, 2024-07-22:
This is very versatile!

However, I am not sure that using 3 different fonts/sizes for the same number is as intuitive as it could be as a default.

When designing the current standard, I've considered the LEGO approach (0--19) and an approach similar to yours (up to 499 in a slightly different way), but ultimately settled on a 0-99 approach as a balance between simplicity and functionality.

Note that if you want "full size" characters for 0--9, you can already do so with the `char` method:

```python
hub.display.char(str(3))
```

But I think you've found those already:

> I found a bit wacky and hard to read.

These use the standard MicroBit font, but I would be happy to take a pull request to improve them. A centered 3x5 font is fine by me. It should be as simple as adapting `pybricks/common/pb_type_lightmatrix_fonts.c`:

[image]

------

Making a library for additional numbers like you have seems like a good approach to enable bigger numbers.

We can definitely add some generic hooks that might make your library usage easier.  In fact, maybe there already is something for you....

-------

As you could see above, Pybricks icons are encoded in a single 32-bit integer. There is an undocumented feature that lets you make icons like those easily:

```python
HEART = Icon(11533764)
```

And since you can add matrices like these, you can fairly easily build composite numbers from just 20 integers in your code, which should make the library very light. Most of the math will happen at the C level too, so it should be fast.

While we're on the topic, we would welcome additional default icons in `pybricks/parameters/pb_type_icon.c` too. With each icon taking just one `qstr` and one 32-bit int, there isn't really a limit to what we can add.

When we do, we could also consider adding a corresponding Icon dropdown + field in the block editor. We haven't really leveraged that functionality so far, even though Pybricks does make animations pretty powerful and easy.

afarago, 2024-07-22:
Awesome!

I started working on the micropython codebase to implement this.
Quick question: I will need a method to display a matrix with two different brightness (e.g. 50 and 100).

Neither pbio_light_matrix_set_rows nor pbio_light_matrix_set_image suits this.
Definitely the uint8_t[5] bit format would be preferred.

My best guess: I would extend the pbio_light_matrix_set_rows function to this 

pbio_error_t pbio_light_matrix_set_rows(pbio_light_matrix_t *light_matrix, const uint8_t *rows, const uint8_t multiplier = 100);

That I would use like this for displaying "42".
```cpp
pbio_light_matrix_set_rows(light_matrix, [0b20211,0b20201,0b22211,0b00210,0b00211], 50);
// [0b101,0b101,0b111,0b001,0b001] // 4
// [0b11,0b01,0b11,0b10,0b11] // 2
// [0b20211,0b20201,0b22211,0b00210,0b00211] // 42
```

Does this sound alright with you? May I continue like this?

laurensvalk, 2024-07-22:
I thought I saw a comment here just now but now I can't find it. In any case, to get half brightness you can do:

Icon(11533764) * 0.5.

I'm simplifying and not covering edge cases, but your library function could be something like:

```
return Icon(left_digits[...]) * 0.5 + Icon(right_digits[number % 10])
```

afarago, 2024-07-22:
Awesome!

I started working on the micropython codebase to implement this.
Quick question: I will need a method to display a matrix with two different brightness (e.g. 50 and 100).

Neither `pbio_light_matrix_set_rows` nor `pbio_light_matrix_set_image` suits this.
Definitely the uint8_t[5] bit format would be preferred.

My best guess: I would extend the `pbio_light_matrix_set_rows` function to this:

`pbio_error_t pbio_light_matrix_set_rows(pbio_light_matrix_t *light_matrix, const uint32_t *rows, const uint8_t bits_per_pixel = 1, const uint8_t multiplier = 100);`

That I would use like this for displaying "42".
```cpp
pbio_light_matrix_set_rows(light_matrix, [0b1000100101,0b1000100001,0b1010100101,0b0000100100,0b0000100101], 2, 50);
// [0b101,0b101,0b111,0b001,0b001] // 4
// [0b11,0b01,0b11,0b10,0b11] // 2
// [0b1000100101,0b1000100001,0b1010100101,0b0000100100,0b0000100101] // 42
```

Does this sound alright with you? May I continue like this?

Alternative approach with kwargs:

```cpp
pbio_light_matrix_set_rows(light_matrix, [0b101,0b101,0b111,0b001,0b001], 100, [0b11,0b01,0b11,0b10,0b11], 50);
// [0b101,0b101,0b111,0b001,0b001] // 4
// [0b11,0b01,0b11,0b10,0b11] // 2
```

laurensvalk, 2024-07-22:
Ah, the comment is back now :slightly_smiling_face: 

> I started working on the micropython codebase to implement this.

Oh, I meant the `pybricks-micropython` pull request mainly for font tweaks.

To show digits with reduced brightness, [multiply with a scalar](https://github.com/pybricks/support/issues/1711#issuecomment-2243270735).

afarago, 2024-08-23:
As a next step would you support to differentiate the single digit (0 - 9 and -9 - -1 range) for better readibility in `hub.display.number()` -> I create a PR
or you rather prefer consistency between single and double digit numbers? -> I close this issue

laurensvalk, 2024-08-26:
Since you have already improved the character displays for 0--9 (thanks for https://github.com/pybricks/pybricks-micropython/pull/253!), I think we can keep `display.number` consistent as it is.

If you used `display.number` to display sensor values from 0--100 for example, you often keep your eye on the most significant number on the left. It would be odd if it disappeared for small numbers.

If you prefer the different font for smaller numbers, you can always do that in your code.

closed 2024-08-26 by laurensvalk