pybricks/support #1723: Pybricks suddenly stopped working

NinjaAdam, 2024-07-25:
I am trying to put code on my Spike Prime Hub using Pybricksdev to put Code/Main.py on it but I am now getting this error

Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\user\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\user\.local\bin\pybricksdev.exe\__main__.py", line 7, in <module>
  File "C:\Users\user\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\cli\__init__.py", line 389, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File "C:\Users\user\AppData\Local\Programs\Python\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\user\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "C:\Users\user\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\cli\__init__.py", line 204, in run
    await hub.run(script_path, args.wait)
  File "C:\Users\user\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\connections\pybricks.py", line 572, in run
    mpy = await compile_multi_file(py_path, abi)
  File "C:\Users\user\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\compile.py", line 141, in compile_multi_file
    mpy = await compile_file(relative_file, abi_major)
  File "C:\Users\user\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\compile.py", line 53, in compile_file
    with open(path, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'Main.py'

laurensvalk, 2024-07-25:
Which command did you use to run it? If your main program called `Main.py`?

NinjaAdam, 2024-07-25:
> Which command did you use to run it? If your main program called `Main.py`?

I used `pybricksdev run ble Code/Main.py`

laurensvalk, 2024-07-25:
Can you navigate into the `Code` folder and try the following?

`pybricksdev run ble Main.py`

NinjaAdam, 2024-07-25:
> Can you navigate into the `Code` folder and try the following?
> 
> `pybricksdev run ble Main.py`

That worked!

closed 2024-07-25 by NinjaAdam