pybricks/support #1743: Pybricks Code hub stdout to terminal is not UTF-8 aware

dlech, 2024-08-06:
### Discussed in https://github.com/orgs/pybricks/discussions/1742

<div type='discussions-op-text'>

<sup>Originally posted by **angzicn** August  6, 2024</sup>
Thank you, I'm a Chinese pybricks fan and I want to teach my son use pybricks to make python code.
Can i print the Chinese str in the online terminal?
There is something wrong with the Chinese str printing in the terminal.
incomplete display,  like this↓
[image]
thank you！</div>

dlech, 2024-08-06:
The problem is that data sent from the hub gets split up into 20 byte packets. If a multi-byte character gets split, Pybricks code currently isn't catching that and print two or more invalid characters instead of recombining the data into whole characters.

closed 2024-09-08 by dlech

BertLindeman, 2024-10-15:
beta.pybricks.com has just been updated and now printing utf-8 characters look nice!

Small example:
[image]

So fixed!

Bert

laurensvalk, 2024-10-15:
I was going to add an example for @angzicn to say 很高兴认识你 (我学习中文) but bonus points for extra effort go to @BertLindeman :smile: Love it - thanks for sharing! I suppose you can make an animated story now. I think we support \r. Is there a sequence to go back a line?

BertLindeman, 2024-10-15:
Animated, I do not know, but I can share the program (need to cleanup all debugging)
Will do a discussions "look what I made item"

There are domino-stone images that could be animated. The \r worked before this update and I did not yet try it on this beta.

dlech, 2024-10-15:
> Is there a sequence to go back a line?

And much, much more: https://xtermjs.org/docs/api/vtfeatures/

BertLindeman, 2024-10-15:
Misunderstood you Laurens, you were just stating the fact, not asking me to...., 😄 

>  suppose you can make an animated story now

I use ANSI codes as an example to show xbox controls sent by an disconnected Technichub on some other hub.
The controls stay at their location for better understanding.
(Horrible code though)
 
Added the UTF-8 print program in discussion #1882 .