pybricks/support #1751: Pybricksdev regularly gets stuck while uploading program

thomasbrus, 2024-08-10:
Well first of all, for PacketLogger to work on Mac OS X one needs to install a Bluetooth logging profile now so that took me a little while to figure out.

> The Bluetooth logging profile is now required on macOS to use PacketLogger (same as iOS). This first started in recent macOS 14.5 seeds.

> Please install the Bluetooth logging profile from: https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=macos&name=bluetooth and try this again.

https://forums.developer.apple.com/forums/thread/750801?answerId=786523022#786523022

I'm on `Mac OS X (intel Sonoma 14.5)`, using pybricksdev version ` v1.0.0a50` and Python `3.10.14`. It regularly happens happens that it gets stuck uploading (` pybricksdev run ble main.py --name "<hub name>"`). The larger the program the more often it seems to occur (makes sense since they take longer to upload I suppose).

It doesn't seem to make a difference whether I turn off bluetooth on nearby devices or not, it seems quite random. I attached two logs, the first one got stuck at 4% and the other one at 2%, but it also happens further along sometimes. 

[swerve_drive.log](https://github.com/user-attachments/files/16460798/swerve_drive.log)
[swerve_drive_2.log](https://github.com/user-attachments/files/16460828/swerve_drive_2.log)

thomasbrus, 2024-08-11:
Ok, do let me know if you need any other information

thomasbrus, 2024-09-29:
@dlech it works a lot better already now that I have an M3 macbook, no issues at all so far.

grawuli, 2024-11-01:
> 
>  It looks like we have a bug in the firmware.
> 

Hey David,

is this Bug fixed?

cu, Markus