pybricks/support #1763: [Bug] Enable python compile on iPad/iPhone for V Browser

afarago, 2024-08-16:
**Describe the bug**
iPhone browser "V Browser" fails to compile template app.
While native iPhone browser does not support web Bluetooth, V browser does support it and can successfully connect to Pybricks hub.

It fails on starting any programs though.

**To reproduce**
Steps to reproduce the behavior:
1. Download and start V Browser
2. Click on Connect Bluetooth button and select a hub
3. Use the plus sign to create a template program.
4. Press play
5. See the error

**Expected behavior**
What did you expect to happen instead?
Program compiles and gets downloaded.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[image]

See #1722 for standard browser behaviour on iPad.

dlech, 2024-08-16:
It looks like this browser doesn't implement the require Web Bluetooth functions. I would try a different browser.

This one is [open source and a fan of Pybricks](https://github.com/daphtdazz/WebBLE/pull/59): https://apps.apple.com/gb/app/webble/id1193531073

I've also seen this one - don't know if it has the required APIs:  https://apps.apple.com/us/app/bluefy-web-ble-browser/id1492822055

afarago, 2024-08-17:
The V browser is the only free one I have found that is able to connect to the Hub, and it is also able to successfully flash some of my othre BLE projects.

I see now that BluetoothRemoteGATTCharacteristic.writeValueWithResponse() is an experimental API, so now I understand why V Browser does not implement it yet.

Based on this info I can close this ticket. Thanks for adding the context and the links.

triplef, 2025-04-20:
> Some possible alternatives (I don't have an iPad, so I've never tried any of these)...

I've tested these to use Pybricks on an iPad. 

> This one is [open source and a fan of Pybricks](https://github.com/daphtdazz/WebBLE/pull/59): https://apps.apple.com/gb/app/webble/id1193531073

WebBLE generally works, except for the issue in #1963. 

> I've also seen this one - don't know if it has the required APIs:  https://apps.apple.com/us/app/bluefy-web-ble-browser/id1492822055

This showed an error for me when trying to connect the hub via Bluetooth.

afarago, 2025-08-24:
I did some coding and testing locally I was able to make it work. Feeling quite stupid I have not tried it back in 2024.

The [writeValue](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/writeValue) is deprecated, and [writeValueWithResponse](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/writeValueWithResponse), [writeValueWithoutResponse](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/writeValueWithoutResponse) is the recommended functions from webBluetooth.

Actions needed:
* polyfill for older browsers to fallback to writeValue
* include the polyfill in an early stage (e.g. index.ts)

Works fine with the [V Browser](https://apps.apple.com/us/app/the-v-browser/id6446792641) on iPhone.

Let me know if I should create a PR for this.

[bleWritePolyfill.ts](https://github.com/user-attachments/files/21958557/bleWritePolyfill.ts)

dlech, 2025-08-24:
Did you try flashing firmware on all hubs? The bootloader can crash if the wrong write type is used depending on the command. And the characteristics report the incorrect properties for what is actually supported.

Fun fact: Pybricks is the [reason](https://chromestatus.com/feature/5152041850634240) `writeValue()` is deprecated. It just didn't work with the LEGO hubs, so we added the two "new" methods (added 5 years ago, so not really new at all).

I guess it doesn't hurt to have a polyfill, but it would be best if the browsers actually added these methods.

afarago, 2025-08-24:
I did programming on already flashed SPIKE Prime/RI hubs.

Primarily I would see this upgrade to benefit the programming on iPad/iPhone/iPad.

Flashing though:
* Move and Technic hub - would be able to try the flashing, I have those
* SPIKE - no BT flashing
* Spike Essential - no access to such device

Congrats on the API change writeValue() --> great job :D

afarago, 2025-08-24:
Just tried it with a Technic Hub and works like a charm...
[image]