pybricks/support #1771: [Bug] Motor Angle Position Problem

fmorton, 2024-08-22:
Using the block interface, when positioning the motors to a specified angle, it moves "close" to the angle and then frequently has a small adjustment. Makes it difficult to do things musically, like a xylophone player. Doing the same thing with the Lego tools positions precisely and quickly.

See code and results in these two videos, one with pybricks and one with the Lego tools.

https://www.youtube.com/shorts/C0HYBQYLoFo

https://www.youtube.com/shorts/Lfl4z2hCgsc

Any way around this problem?

BertLindeman, 2024-08-22:
Hello Frank,

I do not know about the how and what, but I tried this program and I do not see the small movements after the motor moves. 

[image]

How would hat work for you?

Bert

laurensvalk, 2024-08-23:
Hi @fmorton!

> when positioning the motors to a specified angle, it moves "close" to the angle and then frequently has a small adjustment.

This happens with the default `hold` mode, where it will readjust continuously. This can be helpful to keep mechanisms in place even under load. But you don't really need it if there isn't any load.

The LEGO app also uses the Pybricks motor drivers, but the default stop mode is `brake`. You can use `brake` as in Berts example above to get the equivalent result.

fmorton, 2024-08-26:
No surprise, but you two are exactly right. This is not a bug, just me learning. Your suggestion worked perfectly.

laurensvalk, 2024-09-16:
Glad it worked! I'll close this issue since the question was answered.

closed 2024-09-16 by laurensvalk