pybricks/support #1775: [Bug] parallel await wait() never finishes

dlech, 2024-08-24:
**Describe the bug**
Calling multiple `await wait()` in parallel tasks causes tasks to lock up.

**To reproduce**

```python
from pybricks.tools import wait, run_task, multitask, StopWatch

def count():
    n = 0

    while True:
        yield n
        n += 1

async def one():
    timer = StopWatch()

    for c in count():
        if c % 100 == 199:
            print(c / timer.time() * 1000)

        await wait(5)

async def two():
    while True:
        await wait(1)

async def main():
    await multitask(one(), two())

run_task(main(), loop_time=0)
```

Running program produces no output.

**Expected behavior**

`await wait(0)` should act as a yield statement to allow other task to run.

Program should print `5` every 1/2 second.

dlech, 2024-08-24:
More weird behavior - any `loop_time` < 5 results in ~5ms delay.

dlech, 2024-08-24:
In a program that does more in the loop, the delay for `await wait(1)` is always ~10.5 ms instead of ~5.

```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor
from pybricks.parameters import Port, Direction
from pybricks.tools import wait, StopWatch, run_task, multitask
from pybricks.geometry import Axis

# Initialize motors.
left = Motor(Port.C, Direction.COUNTERCLOCKWISE)
right = Motor(Port.D)
left.reset_angle(0)
right.reset_angle(0)

# Initialize hub and color sensor.
# You can also use the TechnicHub and/or the ColorDistanceSensor
# instead.
hub = InventorHub()
sensor = ColorSensor(Port.A)

def count():
    n = 0
    while True:
        yield n
        n += 1

async def balance():
    # Initialize position buffer for speed calculation.
    DT = 5
    window = 300 // DT
    buf = [0] * window
    idx = 0
    angle = 0
    DRIVE_SPEED = 0
    PAUSE = 5000

    # Timer to generate reference position.
    watch = StopWatch()
    counter = count()

    while True:
        # Get angular rate and estimated angle.
        rate = hub.imu.angular_velocity(Axis.Y)
        angle += rate * DT / 1000

        # Get motor position.
        position = (left.angle() + right.angle()) / 2

        # Calculate motor speed.
        speed = (position - buf[idx]) / (window * DT) * 1000
        buf[idx] = position
        idx = (idx + 1) % window

        # Calculate reference position, which just grows linearly with
        # time.
        reference = -max(watch.time() - PAUSE, 0) / 1000 * DRIVE_SPEED

        # Calculate duty cycle.
        diff = position - reference
        duty = 0.018 * rate + 19 * angle + 0.45 * diff + 0.16 * speed

        # Account for battery level and type.
        duty *= 7200 / hub.battery.voltage()

        # Calculate steering.
        # reflection = await sensor.reflection()
        # steering = (reflection - 28) * 0.6
        steering = 0

        # Apply duty cycle for balancing and steering.
        left.dc(duty + steering)
        right.dc(duty - steering)

        c = next(counter)
        if c % 500 == 499:
            print(watch.time() / c)

        await wait(1)
        
# async def sound():
#     # TODO - replace with actual code
#     while True:
#         await wait(1)

# async def main():
#     await multitask(balance(), sound())

run_task(balance(), 5)
# run_task(main())
```

dlech, 2024-08-24:
I think the problem is that any awaitable that returns on the first iteration doesn't yield. So basically, 

```python
while True:
    await wait(0)
```

is the same as

```python
while True:
    pass
```

The behavior of `asyncio.sleep(0)` in CPython is:

> Setting the delay to 0 provides an optimized path to allow other tasks to run. This can be used by long-running functions to avoid blocking the event loop for the full duration of the function call.

So I think we should do the same with `wait(0)`.

laurensvalk, 2024-08-26:
This is a duplicate of https://github.com/pybricks/support/issues/1775

dlech, 2024-08-26:
> This is a duplicate of #1775

Duplicate of itself? :smile:

dlech, 2024-08-26:
Ah here is the actual duplicate: https://github.com/pybricks/support/issues/1429

laurensvalk, 2024-08-26:
:smile:

BertLindeman, 2024-08-26:
Confirm primehub at build 3503 printing where build 3501 did not print.

```python

# prints nothing on build 3501 

from pybricks.tools import multitask, run_task, wait

async def main1():
    while True:
        await wait(0)

async def main2():
    while True:
        print('Hello!')
        await wait(500)

async def main():
    await multitask(main1(), main2())

run_task(main())
```

closed 2024-08-28 by laurensvalk