pybricks/support #1787: GATT operation failed for unknown reason.

AlanMurray350, 2024-08-28:
An unexpected error occurred. Please consider reporting this so we can make a better error message.

GATT operation failed for unknown reason.

Trying to install firmware ...

laurensvalk, 2024-08-28:
Which hub, browser, and operating system are you using?

patmuk, 2024-09-08:
I have this error as well:
```
GATT operation failed for unknown reason.
```
MacOS 14.6.1., M3 Max
Chrome Version 127.0.6533.89 (Official Build) (arm64)

Steps are
1. Erasing old firmware (this is very fast)
2. Uploading new firmware 
3. 1% ... 2% ... 3% ... 4%
and then the above error message comes.

laurensvalk, 2024-09-08:
Which hub are you using?

dlech, 2024-09-08:
We've had some reports of other folks having problems with newer Macs as well. Could you log Bluetooth packets and share the results as described in #270?

patmuk, 2024-09-08:
> Which hub are you using?

Lego Boost (sorry for forgetting the most important information!)

patmuk, 2024-09-08:
> We've had some reports of other folks having problems with newer Macs as well. Could you log Bluetooth packets and share the results as described in #270?

Thanks! Will do (later) and post here.

patmuk, 2024-09-08:
This is the chrome log:
````
BluetoothEvent[2024/09/08 19:04:03.736878] 
Classic device added: 00:16:53:A4:70:4B service UUIDs:

BluetoothEvent[2024/09/08 19:04:03.736664] 
Device connected: name: LEGO Bootloader address: 00:16:53:A4:70:4B

BluetoothEvent[2024/09/08 19:04:00.298699] 
Web Bluetooth Device Chooser initiating Bluetooth discovery session
``` 

Looks like the connection cannot be established stable?

dlech, 2024-09-08:
Chrome log isn't very useful. We need actual Bluetooth packet logs.

https://bleak.readthedocs.io/en/latest/troubleshooting.html#macos

patmuk, 2024-09-08:
Sure, here is the bluetooth log from `PacketLogger.app`
I filtered it for the Device "00:16:53:A4:70:4B", which I believe the Lego Boost Hub is (as this appeared when starting),
<details>
  <summary>Log</summary>
``` 
Sep 08 20:43:53.112  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -71 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:53.112  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -71 dBm - LEGO Bootloader - Channel 37  
Sep 08 20:43:53.395  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -67 dBm - Manufacturer Specific Data - Channel 38  
Sep 08 20:43:53.395  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -68 dBm - LEGO Bootloader - Channel 38  
Sep 08 20:43:53.742  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Low Power - Public - 00:16:53:A4:70:4B  -71 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:54.023  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -71 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:54.025  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -72 dBm - LEGO Bootloader - Channel 37  
Sep 08 20:43:54.301  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -67 dBm - Manufacturer Specific Data - Channel 38  
Sep 08 20:43:54.581  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -77 dBm - Manufacturer Specific Data - Channel 39  
Sep 08 20:43:54.582  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -77 dBm - LEGO Bootloader - Channel 39  
Sep 08 20:43:54.897  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -69 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:54.930  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -69 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:54.932  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -69 dBm - LEGO Bootloader - Channel 37  
Sep 08 20:43:55.177  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -67 dBm - Manufacturer Specific Data - Channel 38  
Sep 08 20:43:55.178  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -67 dBm - LEGO Bootloader - Channel 38  
Sep 08 20:43:55.492  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -75 dBm - Manufacturer Specific Data - Channel 39  
Sep 08 20:43:55.492  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -75 dBm - LEGO Bootloader - Channel 39  
Sep 08 20:43:55.842  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Low Power - Public - 00:16:53:A4:70:4B  -68 dBm - Manufacturer Specific Data - Channel 38  
Sep 08 20:43:55.842  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Low Power - Public - 00:16:53:A4:70:4B  -67 dBm - LEGO Bootloader - Channel 38  
Sep 08 20:43:56.086  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -68 dBm - Manufacturer Specific Data - Channel 38  
Sep 08 20:43:56.086  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -67 dBm - LEGO Bootloader - Channel 38  
Sep 08 20:43:56.401  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -82 dBm - Manufacturer Specific Data - Channel 39  
Sep 08 20:43:56.401  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -82 dBm - LEGO Bootloader - Channel 39  
Sep 08 20:43:56.436  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -82 dBm - Manufacturer Specific Data - Channel 39  
Sep 08 20:43:56.436  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -81 dBm - LEGO Bootloader - Channel 39  
Sep 08 20:43:56.679  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -70 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:56.680  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -70 dBm - LEGO Bootloader - Channel 37  
Sep 08 20:43:56.958  HCI Command      0x0000  00:16:53:A4:70:4B  LE Add Device To Filter Accept List - Public - 00:16:53:A4:70:4B  
Sep 08 20:43:56.998  HCI Event        0x0000  00:16:53:A4:70:4B  LE - Ext ADV - 1 Report - Normal - Public - 00:16:53:A4:70:4B  -69 dBm - Manufacturer Specific Data - Channel 37  
Sep 08 20:43:56.998  HCI Event        0x0050  00:16:53:A4:70:4B  LE - Enhanced Connection Complete - Central - Public - 00:16:53:A4:70:4B local - 00:00:00:00:00:00 peer - 00:00:00:00:00:00 - Conn Interval: 30 ms  
[... cut ...]
Sep 08 20:44:03.173  ATT Send         0x0050  00:16:53:A4:70:4B  Write Command - Handle:0x000E - 00001626-1212-EFDE-1623-785FEABCD123 - Value: 2212 5A60 0008 013C 2370 E5E7 F0B5 0600…  
Sep 08 20:44:03.173  ATT Send         0x0050  00:16:53:A4:70:4B  Write Request - Handle:0x000E - 00001626-1212-EFDE-1623-785FEABCD123 - Value: 66  
Sep 08 20:44:03.200  HCI Event        0x0050  00:16:53:A4:70:4B  Number Of Completed Packets - Handle: 0x0050 - Packets: 0x0001    
Sep 08 20:44:03.200  HCI Event        0x0050  00:16:53:A4:70:4B  Disconnection Complete - Success  
```  
</details>

dlech, 2024-09-08:
Thanks. You can just put it in a zip file, then we can open it with Wireshark.

patmuk, 2024-09-08:
Thanks - I attached the file in "pklg" format (btsnoop) was possible as well. I did not record it with Wireshark, I used PacketLogger.app - hope that was correct.

[LegoBoost_on_m3.pklg.zip](https://github.com/user-attachments/files/16923431/LegoBoost_on_m3.pklg.zip)

dlech, 2024-09-08:
Thanks for the logs. 

I was actually able to reproduce the issue on my Mac. I think the problem is that macOS is just too efficient and is causing Bluetooth packets to be sent faster than the hub can handle them.

So we probably just need to slow down the firmware update a bit in Pybricks Code. 

Interestingly, the firmware update works fine in a development build, but fails in a production build.

In your logs, we can see there is just a few milliseconds between the first and last Write Request in a burst.

[image]

Compared to the working development build where there is closer to 10 milliseconds per burst.

[image]

closed 2024-09-08 by dlech

patmuk, 2024-09-09:
Wow, thanks for handling this issue so fast (analysis, responses and code fix!).
Are you CDing in the web app, or when will the release be live?

laurensvalk, 2024-09-09:
>  I think the problem is that macOS is just too efficient and is causing Bluetooth packets to be sent faster than the hub can handle them.
> 
> So we probably just need to slow down the firmware update a bit in Pybricks Code.

I was having a bit of deja vu reading that.... :smile: You are very consistent:

> So most likely, your Bluetooth on this particular computer is so efficient that it sent data to the hub faster than the hub could process it, causing the hub to lock up.
>
> So we probably need to slow down firmware updates a bit to make sure this can't happen.

_Originally posted by @dlech in https://github.com/pybricks/support/issues/1583#issuecomment-2052610621_
            
It sounds like that issue can now be closed too?

> Are you CDing in the web app, or when will the release be live?

We release beta versions a bit more frequently. I think we'll do one in a few weeks which will include this fix. See [this link](https://pybricks.com/learn/getting-started/install-pybricks/#trying-new-beta-features-optional ) for more details.

In the mean time, perhaps you could try doing the firmware installation from another device, and then return to your Mac for regular coding? If you have any Android device, that's usually the easiest.

dlech, 2024-09-09:
No, loading firmware works different from loading programs. If you want to start a new issue an include Bluetooth packet logs of a failed download, we can have a look.

BertLindeman, 2024-09-09:
I cannot get a solid situation in this. Sometimes I just need to reboot windows.

Will keep a trace in mind.

Thanks David.