pybricks/support #1799: Program slots and Pybricks Code

laurensvalk, 2024-09-03:
> One thing I not so sure about though is how to deal with multiple "user" slots in Pybricks Code.
> 
> It would probably be a good idea to add some kind of way to read which slots have a valid program in them.
> 
> With WebBluetooth, we can't get the actual error code for failures. So if we attempt to run a program in a slot with a bad program, there is no way to know why the error occurred. Was it out of range? Was it a bad program? Did the hub battery die? Right now, we don't have any "expected" errors, so we kind of ignore this problem.
> 
> So that's why I'm thinking it would be better to have something where we can "ask first" before trying something that would fail, so that we don't actually ever try to do something that might fail under normal circumstances.

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/264

laurensvalk, 2024-09-03:
We'll also want to tell Pybricks Code the maximum available size for the current slot, since it is no longer the fixed size we provide in the hub capabilities.

laurensvalk, 2024-09-17:
Gathering some notes from elsewhere in one place:

> At minimum, we'll also want to do the following:
> - [ ] Introduce a status flag such as `PBIO_PYBRICKS_STATUS_RECEIVING_PROGRAM` to prevent the UI interactions like a slot change while a big program is being loaded. This isn't done yet, so changing slot while loading lead to failure.
> - [ ] Inform Pybricks Code about the maximum size for the current slot, or all slots. Perhaps also include the slot in the event status report (slot is not necessarily the same as program ID). Right now, Pybricks Code won't know that storage is full until it is actually full.
> 
> We can probably rethink the logic for downloading programs more generally. At the moment, Pybricks Code sets the program size to 0 (to prevent it from being run), then uploads the program, then downloads the size. This is no longer sufficient, hence the need for a status. Alternately, maybe this should be a process of some kind so the hub can know about failure (and clear that status). Right now, if uploading stops halfway the hub has no way of knowing since it doesn't know the final file size.
> 
_Originally posted by @laurensvalk in https://github.com/pybricks/pybricks-micropython/pull/266_

> I would much prefer to see slots saved to external flash when not in use rather than trying to keep all slots in RAM at the same time. Otherwise, we are wasting a lot of RAM.

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/266_

BertLindeman, 2024-12-18:
I used slots for a few  days.
Works nice. 
I had in slot five a program to show memory usage.
So load a program in slot 1..4 and run slot 5 to see the diff :smile: 
 
Each time the hub restarts, it defaults to slot 1.
This caused me to accidentally overwrite slot 1 when I intended to continue with a program in e.g. slot 3.
(my memory . . . . )

Would it be a good idea (and feasible) to make the hub remember the last slot used and set it as the default on the next start?