pybricks/support #1800: [Feature] Allow scanning for remote asynchronously.

laurensvalk, 2024-09-03:
> I was wondering if it would be possible to connect asynchronously with the controller at any time while the main program is already running, and let the robot check whether the controller is connected and behave differently when it is (remote-controlled) or not (autonomous).

_Originally posted by @DanieleBenedettelli in https://github.com/pybricks/support/discussions/1598_

laurensvalk, 2024-09-03:
> What I think would be helpful is if the XboxController constructor took an `auto_connect` parameter.  So you could create `XboxController(auto_connect=False)`, and then later you could call an (async) `connect` method.  When the controller isn't connected then it should behave as if the joysticks are centered and nothing is pressed.
>
> That way you could write code for menus, for example, that could use either the controller or the hub buttons to control the robot.  With just a connection timeout parameter, I think we'd have to create a wrapper class around the controller instead of using it directly.
>
> It seems like an `auto_connect` parameter and `connect` method might also work well with the block coding.

_Originally posted by @dsplaisted in https://github.com/pybricks/support/discussions/1598#discussioncomment-9166479_

laurensvalk, 2024-09-03:
This would also be nice to have for the upcoming builtin "port view" or "port control" application discussed in https://github.com/pybricks/support/issues/139

Then we can have it scan for remotes, but also allow you to operate it without a remote.

We can make make this optional, and make it use the current behavior as the default so that it isn't a breaking change.

abliss, 2025-07-29:
Just want to add: in the current version of Blocks, you have to connect to the controller in Setup, which means it's impossible to do anything first to give the user some instruction (like play a sound, print to the console, or display a special character); you're just looking at the default "program running spinner" and left wondering why nothing is happening.  

An alternate solution might be to  just remove the "controller" setup block entirely and have the controller connect automatically the first time a block tries to access it. Or, I suppose, add the ability to run arbitrary blocks during the Setup stack?

abliss, 2025-07-31:
Wow, that is amazing! I stand totally corrected (and once again, even more impressed by pybricks). Thanks, @laurensvalk  !