pybricks/support #1804: [Bug] Chrome out of memory on large programs

laurensvalk, 2024-09-05:
While creating a few big programs to test the limits of our hub file storage, I noticed that Pybricks Code can crash Chrome. In one case, it dropped into the debugger before the potential memory issue. In other cases, Chrome crashes with `SIGILL`.

This is a bit of an unusual program, but not entirely out of the ordinary. If someone wants to include a binary blob to include some extra data, a program may look not unlike the one below.

Example. 

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor, ForceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = PrimeHub()

data = b"132456789132456789132456789132456789132456789132456789132456789132456789"
data = b"132456789132456789132456789132456789132456789132456789132456789132456789"
data = b"132456789132456789132456789132456789132456789132456789132456789132456789"

# Repeat this MANY more times. Too big for a GitHub post. See attachment for whole file.

data = b"132456789132456789132456789132456789132456789132456789132456789132456789"
data = b"132456789132456789132456789132456789132456789132456789132456789132456789"
data = b"132456789132456789132456789132456789132456789132456789132456789132456789"
data = b"132456789132456789132456789132456789132456789132456789132456789132456789"

hub.light.on(Color.BLUE)
wait(10000)
```

dlech, 2024-09-08:
This seems to be an issue with running the module resolver in Python via Pyodide. I tried updating to the latest Pyodide, but it still crashes.

dlech, 2024-09-08:
Apparently, this is caused by the Python parser that we are using to find resolve imports.

https://github.com/pybricks/python-program-analysis

ggramlich, 2024-10-03:
@laurensvalk One of my boys uses Chrome and when trying to open code.pybricks.com, the browser and also screen recording software crashes after a few seconds, so I could not record a video.
I already tried to delete cookies and caches.
In incognito mode he can use pybricks.
I think that the programs are still in stored in chrome and that causes the problem, but I don't know how to delete them.
Might this be related with this bug? Can you help us deleting the programs, if that might cause the problem?

laurensvalk, 2024-10-03:
Thanks for reporting. I watched your video. Very unusual.

> I think that the programs are still in stored in chrome and that causes the problem, but I don't know how to delete them.

From your video, you can see clear site data. That should do it.

[image]

ggramlich, 2024-10-03:
Hi Laurens,

thanks for the help. We were able to clear the data. But that did not
actually help.
Chrome still crashes.
https://photos.app.goo.gl/B4YsQY9EwGgTZ3KBA

We will probably try to uninstall Chrome then.
Do you have any other idea.

Thanks
Gregor

laurensvalk, 2024-10-03:
Is there anything left in local storage?

[image]

Or IndexedDB > pybricks.fileStorage > contents

[image]

laurensvalk, 2024-10-03:
Does anything show in the console?

ggramlich, 2024-10-05:
@laurensvalk Unfortunately I was not able to dig deeper into it and probably won't be able to do so within the next week as the boy has his laptop with him.
The boy now uses the beta version, which does not cause the problem (neither does incognito mode).
As you saw in the videos, you only have about one second to react after opening the page, so opening a specific node in the Indexed DB tree will be close to impossible. But I will try to see, what I can find in the file system under AppData\Local\Google\Chrome\User Data\Default\Local Storage

ggramlich, 2024-10-13:
@laurensvalk We were able to look at the console output. It showed only the first output "starting Pyodide..." before Chrome crashed.
Then I was able to locate the IndexedDB Folder under AppData\Local\Google\Chrome\User Data\IndexedDB\
The folder for code.pybricks.com showed 4 GB (3.73 GiB) Disk Usage. After deleting the folder, it is possible to use pybricks again.

[image]

laurensvalk, 2024-10-15:
Were you able to create a backup of the folder before deleting it, by any chance?

ggramlich, 2024-10-15:
Sorry - I don't have a backup. I wanted to copy the folder over to our Google Drive shared folder, but the 4 GB were too much. I could have used a USB Stick but we were in our practice under a bit of time pressure.

laurensvalk, 2024-10-15:
No problem at all. Thanks as always for taking the time to report this.

ggramlich, 2025-05-15:
The problem from October 2024 happened again. Pybricks closes immediately after opening.
The IndexedDb Folder was about 5GB and after deleting that folder Pybricks is usable again.
Unfortunately the boy lost his work of the last 10 days. I even told him the day before, when he could still work with Pybricks to make a backup and delete some of the old programs. He didn't want to. Next time he might be more willing to do that.

I now have a 3GB zip file of the IndexedDb folder in my Nextcloud.

@laurensvalk @dlech I can send you the link if you want to take a look at it.

dlech, 2025-05-15:
> Unfortunately the boy lost his work of the last 10 days.

😢 

> I can send you the link to download from my Nextcloud if you want to take a look at it.

I think we already identified the source of the problem so probably don't need it at this time.

I had forgotten about this bug though, so thanks for the reminder.

ggramlich, 2025-05-16:
@dlech @laurensvalk Thanks for the response.

It seems like we were extremely lucky. With some AI help and https://github.com/cions/leveldb-cli/ I was able to take a look at the keys in the LevelDB.
Any attempt to export / dump / compact the whole database failed because of memory usage or took more than 30 minutes. I also tried some other programs but they did not support Chromium's IndexedDB database (idb_cmp1 comparer).

There were several thousands keys, if not millions. But the very first key in the list that was a bit longer contained the program that we were looking for.
It might have been that it was on top of the list, because it was the last program that he worked on or that it was because he renamed it to be the first alphabetically.

I put the value for that key in a text file `prog.py`. The text file had some multibyte encoding which displayed fine, when I did a `cat prog.py`, but could not properly be displayed with `less` or opened in a text editor. Finally I was able to restore the (block-code) program completely so that the boy could import it again into Pybricks.

I will keep the zip file and can send a download link to you.

laurensvalk, 2025-05-16:
Wow, well done! Your students have a really awesome coach 😄

ggramlich, 2025-05-16:
I just saw https://github.com/orgs/pybricks/discussions/2192
Sounds similar

ggramlich, 2025-05-16:
@laurensvalk By the way: I found the exact same value (the program) in some of the first keys that I looked at. So there is some unnecessary redundancy going on that might lead to this gigantic memory usage.

ggramlich, 2026-06-09:
Ah - do you mean a backup exported from Pybricks? Yes the boy has done one last Saturday, maybe  Sunday.

closed 2026-09-21 by laurensvalk