pybricks/support #1806: [Feature] Make it possible to stop observing

laurensvalk, 2024-09-06:
The Technic Hub can sometimes lock up when advertising and scanning at the same time (https://github.com/pybricks/support/issues/1419)

Since there may not be a solution due to the Bluetooth chip limitations, maybe we should have a way to stop and restart observing, just as we have for broadcasting.

Then the user can choose which one to use in their script.

laurensvalk, 2024-10-04:
Two variants:

We can add an optional argument to `observe` which can be `True` or `False` to start or stop observing.

This is somewhat similar to sending `None` to stop broadcasting.

```python
hub.ble.observe(True) # Start observing
wait(1000)
data = hub.ble.observe(123)
hub.ble.observe(False) # Stop observing

```

Or introduce a new separate method:

```python
hub.ble.start_observing(True) # Start observing
wait(1000)
data = hub.ble.observe(123)
hub.ble.start_observing(False) # Stop observing

```

BertLindeman, 2024-10-05:
A test for observe_enable gets stuck and disconnects
My program name ```issue_1419_mar_26_build_3584.py```

Should there be a wait between the broadcast and the observe_enable?

```python
# Running hubC at 3584
# issue_1419_mar_26_build_3584
# adapted to use observe_enable

from pybricks.hubs import TechnicHub
from pybricks.tools import wait, StopWatch
from pybricks.parameters import Color
from urandom import choice

watch = StopWatch()
transmitter = TechnicHub(broadcast_channel=1, observe_channels=[2, 3])

while True:
    watch.reset()

    transmitter.ble.observe_enable(False)
    transmitter.light.on(Color.GREEN)
    transmitter.ble.broadcast([choice([True, False]), choice([True, False])])

    transmitter.light.on(Color.VIOLET)
    transmitter.ble.observe_enable(True)

    transmitter.light.on(Color.RED)
    TriggeredDistributor, TableReadyForTipp = transmitter.ble.observe(3) or [0] * 2

    transmitter.light.on(Color.BLACK)
    if watch.time() < 95:
        print(end='.')
        wait(100 - watch.time())  # make it 100 msec per cycle
    watch.reset()
```

It takes minutes and than the hub-led stays violet and hub disconnects.
Short button press does nothing, long button press does a normal power-off. 

Environment: A primehub is running this program:
```python

from pybricks.hubs import PrimeHub
from pybricks.tools import wait
from urandom import choice

transmitter = PrimeHub(broadcast_channel=2, observe_channels=[1, 3])

while True:
    transmitter.ble.broadcast([choice([True, False])])
    TriggeredDistributor, TableReadyForTipp = transmitter.ble.observe(3) or [0] * 2
    print(TriggeredDistributor, TableReadyForTipp, end="\t")
    wait(0)

```

laurensvalk, 2024-10-05:
I think your example still broadcasts and observes at the same time. The idea of the new method is that you can choose to observe when not broadcasting and vice versa. You could potentially do that on a parallel task.

I’ll put some examples together to illustrate later next week.

closed 2024-10-15 by laurensvalk