pybricks/support #1824: [Bug] ``` Error: Failed to execute 'getPrimaryService' on 'BluetoothRemoteGATTServer': GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.     at D.<computed> (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1397636)     at https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1401567     at m (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1402498)     at h (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1402024)     at l (https://code.pybricks.com/static/js/main.1bf60dfd.js:2:1402281) ```

jxpybrick, 2024-09-16:
**Describe the bug**
What is the problem?

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

retostockli, 2025-12-27:
I have the same issue with both of our two LEGO Boost Hubs:
Tried multiple times with the following Configs:
- Windows 10 & Microsoft Edge & Realtek USB Bluetooth 5.4 dongle
- Linux Ubuntu 24.04 & Chromium & Internal Laptop Bluetooth 
- Samsung Android & Chrome & Samsung Bluetooth
- Samsung Galaxy Tab & Chrome & Samsung Bluetooth

I have no idea what next to try. Obviously it is not related to the type of bluetooth adapter but to the LEGO Boost? Any Ideal?

retostockli, 2025-12-27:
So, this may help do diagnose further.

After failing on all the above devices I've now installed pybricksdev on Linux Ubuntu and tried to load the latest move hub firmewar 3.6.1 via command line. It stops after 1% (as above) with : 

stockli@mariachi:~/Downloads$ pybricksdev flash pybricks-movehub-v3.6.1.zip 
Creating firmware...
Searching for BOOST hub...
Found: 00:16:53:A9:D9:9C: LEGO Bootloader
Connecting to 00:16:53:A9:D9:9C: LEGO Bootloader
Connected successfully!
Erasing flash and starting update
  2%|▊                                        | 1.95k/105k [00:03<02:55, 589B/s]
Traceback (most recent call last):
  File "/home/stockli/.local/bin/pybricksdev", line 7, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/cli/__init__.py", line 620, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/cli/flash.py", line 442, in flash_firmware
    await flash_ble(hub_kind, firmware, metadata)
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/cli/flash.py", line 326, in flash_ble
    await updater.flash(firmware, metadata)
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/flash.py", line 244, in flash
    result = await self.bootloader_request(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/flash.py", line 159, in bootloader_request
    await self.write(data, request.write_with_response)
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/pybricksdev/ble/__init__.py", line 148, in write
    await self.client.write_gatt_char(
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/bleak/__init__.py", line 780, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 841, in write_gatt_char
    assert_reply(reply)
  File "/home/stockli/.local/share/pipx/venvs/pybricksdev/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 27, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error)

retostockli, 2025-12-27:
Another Update: Releasing the Boost Hub button right after the pairing has completed allows to flash the full firmware from the command line on Linux. Keeping the button pressed does not work.

Trying the same trick via the Chromium Web Browser also works. 

So it seems the boost hub button should not be kept pressed after the ereasing / installing x% Firmware has started. Any idea why this differs from the online instructions?

retostockli, 2025-12-27:
Yes, that is correct, thank you. I had completely overseen this message since it only pops up for about 2 seconds until the red error message pops up, about at the 1-2% counter of the firmware update process. So the original post may have had another reason.

retostockli, 2025-12-27:
This could be an idea. Or make the Pop-Up bigger. 

And since there is little time between pairing and start of Firmware Upload, you could possibly modify in step 5:

Keep holding the button in steps 6 and 7 . After pairing, release the button to upload the firmware.