pybricks/support #1839: [Bug] primehub at build 3551 does a shutdown on program download

BertLindeman, 2024-09-18:
**Describe the bug**
Running in beta.pybricks in the installed app. Primehub at build 3551.
Scanario:
1. Connect to hub OK.
2. get warning about 1.4.0 profile OK
3. F5 to run the program shows progress runner without any progress (no percentage)
4. hub silently switches off.
5. The UI gets error `GATT operation failed for unknown reason.` 

**To reproduce**
Steps to reproduce the behavior:
- see above

**Expected behavior**
Hub should run the program.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

[image]

## other data:

**versions**

```Win 11 23H2```

```About Pybricks Beta
Pybricks logo
MicroPython for LEGO® Powered Up smart hubs.

v3.5.0 (Pybricks Beta v2.6.0-beta.2)
```

```Google Chrome Version 128.0.6613.138 (Official Build) (64-bit)```

**Bluetooth trace**
will add later as zip file with two traces
1. original wireshark version
2. current wireshark version

Will debug further trying to see if the program is a factor.

BertLindeman, 2024-09-18:
Can confirm 
1. a user error
2. build 3551  needs extra actions to work

Not sure if this **only** occurs after install of build 3540 (initial slots) 
Should there be something extra in doc or in the firmware?

laurensvalk, 2024-09-18:
Thanks for reporting. This is the same as you've run into earlier. :slightly_smiling_face: 

I've added a commit yesterday (not yet merged) so that it will always delete old user data if the firmware has been changed. See downloads at https://github.com/pybricks/pybricks-micropython/actions/runs/10903733584 if you want to try it sooner.

This was already being done across firmware releases including beta versions, so most users would never run into this. This new commit should take care of developers and active testers like yourself :smile:

BertLindeman, 2024-09-18:
installed build 3540 (first implement slots build) and build 3554 over it. No more problem.

Thanks David.

closed 2024-09-18 by BertLindeman

laurensvalk, 2024-09-18:
And thank you for keeping us on our toes :smile: