pybricks/support #1846: [Bug] Ensure smooth upgrade path to V3.6.0

laurensvalk, 2024-09-25:
We have slightly changed the persistent storage area from V3.6.0b1 onwards. This firmware will also erase the user data if the stored firmware hash does not match that of the current firmware, to prevent issues with future updates. This makes the experience similar to that of the Technic Hub, which erases the data between updates.

But version V3.5.0 did not yet handle deletion in this way, so we'll want to make sure upgrading goes smoothly. If necessary, we can release a version V3.5.1 on the stable channel to prepare.

--------------

> Trying to run (F5) a new program on 3.6.0b1 did the power-off. I did not try to start the "old" program.
> 
> For clarity and re-tested:
> 
> The program:
> 
> ```python
> while True:
>     pass
> ```
> 
> * download 3.5.0 standard via beta.pybricks.com (over 3.6.0b1)
> * (F5) download a new program and run OK
> * download 3.6.0b1 over this 3.5.0
> * pull the USB cable (otherwise I do not see the hub power-off)
> * (F5) download a new program and run results (after ~ a second) in power-off.
>   I see no animation on the display to indicate the program is running
> * then re-install 3.6.0b1 over 3.6.0b1 does not help.

_Originally posted by @BertLindeman in https://github.com/pybricks/support/issues/1577#issuecomment-2029782054_

laurensvalk, 2024-09-25:
Here's why this is happening. If you install 3.6.0b1 and then 3.5.0 and run a program, all seems to work fine, but the saved firmware version is unchanged since 3.5.0 does not erase it. So on 3.5.0 open a REPL and type:

```
>>> hub.system.storage(0, read=8)
b'e75cda81'
```

Which is the version corresponding to [3.6.0b1](https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.0b1). So when you then install 3.6.0b1 again, it doesn't do the erase. But since it does contain a program loaded with 3.5.0, it fails and gets confused.

So to "fix" your hub, @BertLindeman, you can just enter the following on a REPL on version 3.5.0:
```
hub.system.storage(0, write=bytes(8))
```
Then turn off normally and install any version you like.

laurensvalk, 2024-09-25:
So maybe we need a checksum to account for hubs that did not yet store the firmware version, to ensure this isn't an issue when we release a beta version via Pybricks Code in a few weeks.

laurensvalk, 2024-09-25:
Fixed introduced. Switching between the latest [nightly](https://nightly.link/pybricks/pybricks-micropython/workflows/build/master) and 3.5.0 and back again should now work if you run a new program. So once we release this in Pybricks Code beta, everyday users should not be experiencing this issue.

BertLindeman, 2024-09-25:
Just to confirm: install of nightly build over 3.5.0 (and back) does erase the existing program from the hub.

Just did the swap 3.5.0 build 3567. No problem anymore.

laurensvalk, 2024-09-25:
Thanks @BertLindeman, this was an important one, so I'm glad you caught it before we release the beta!

laurensvalk, 2024-12-05:
This is resolved.

closed 2024-12-05 by laurensvalk