pybricks/support #1873: [Bug] Gear train option to Motor() doesn't have any effect

thekatstevens, 2024-10-11:
**Describe the bug**
When configuring multiple motors geared differently, the optional gear train param to the `Motor()` command doesn't seem to be honoured. 

**To reproduce**
Steps to reproduce the behavior:
1. Connect 2 motors (e.g. PUP Large) with a Z16 gear on each, and a third Z16 between them
2. Configure them in PyBricks with the `Motor()` function, using the relevant ports and the same direction
3. Add a gear list to **one** of them, e.g. `[2,1]`
4. Bind them both to a `Car()` and set it up with `car.drive_power()`
5. Run the program and drive the car so both motors turn, and they do so smoothly 

**Expected behavior**
If the code were adjusting power settings automatically based on the gear train, I would have expected the motors to lock up - trying to turn a common gear at two different speeds.

(in practise, I have these motors at the front and rear of a car, but one is geared differently to the other. But when the car drives, the front and rear wheels turn at different speeds, rather than compensating for the specified gear ratio)

laurensvalk, 2024-10-11:
Gearing changes the speed and torque, but not power.

```
power = torque * speed.
```

When one decreases by the gear ratio and the other increases by the gear ratio, power remains the same.

(I'm simplifying here by equating mechanical power and electrical power)

So you could use `drive_speed` instead.  The speed controllers can be a bit stuttery for vehicles. To account for that, increase the `profile` value in the motor setup.

laurensvalk, 2024-12-03:
Closing this as resolved. Thanks!

closed 2024-12-03 by laurensvalk