pybricks/support #1875: [Bug]

crosszay1, 2024-10-11:
**Describe the bug**
got an error upon running the program followed by a bluetooth disconnect and hub shutdown
**To reproduce**
no clue, this has happened to me many times but there does not seem to a cause

**Expected behavior**
my program to run, as a program is supposed to run
**Screenshots**
[image]

please note: right before submitting this I attempted to run the program again, the program ran correctly but yieldedthe same error upon click the stop button

dlech, 2024-10-11:
This error can happen often with certain Bluetooth adapters (particularly built-in Intel Bluetooth on Windows laptops).

If you want to dig further, you can try logging Bluetooth packets while reproducing the problem (if you can).

Info on how to do this and more is in #270.

crosszay1, 2024-10-14:
it seems part of the specs of my low quality laptop have been found..

while it is good to know I am not alone in this matter, it is incredibly frustrating to deal with this issue as it can manifest itself quite frequently. is there any known fixes or at the very least, ways to decrease the volume at which this issue appears?

dlech, 2024-10-14:
You can disable the the built-in Bluetooth and use a USB Bluetooth adapter. There are some ones reported as working well by other users in #921.

crosszay1, 2024-10-15:
I'll try it! thanks for the tip