pybricks/support #1880: Enhance minor inconveniences: bluetooth keep alive and stop the program from CLI

nathanbowang, 2024-10-15:
This is a great project. The graphical programming and no Linux support in Spike software has been really frustrating for me. The Spike robot I bought for my kids has been sitting unused for over a year, and we haven’t started programming together. This project has brought back the joy of programming with LEGO for my family.

According to [this doc](https://pybricks.com/project/pybricks-other-editors/): 

> Pybricksdev will try to connect to the hub, download the program, and start it. It disconnects when the code completes. This approach has some minor inconveniences that you should be aware of:
> 
> * Since it doesn’t stay connected, the hub will turn off after being idle for a few minutes. Just turn it on again with the button.
> * You cannot stop the program from within Visual Studio Code, so you’ll have to use the hub button to stop it.

I believe these two issues can be solved technically, but I wanted to hear if there are other considerations. I would be honored to contribute if possible.

laurensvalk, 2024-10-15:
> This project has brought back the joy of programming with LEGO for my family.

Thank you. That's what we're here for!

> Pybricksdev will try to connect to the hub, download the program, and start it. It disconnects when the code completes.

`pybricksdev` was originally created for the Pybricks developers to conveniently automate testing, including the download process itself. This is why it has the form it has today. It wasn't originally designed to be equivalent to the intended user experience of [Pybricks Code](https://code.pybricks.com/).

We can see the benefit of using your own editor though, and this is why we've still decided published the package and that tutorial to make it work, despite the limitations.

Maybe it makes sense to come up with something different. Either another Python CLI tool or a full fledged VSCode extension that mimics the interactivity of Pybricks Code, with a start/stop/repl button.

Due to other ongoing developments, this hasn't been the highest priority on our list.

nathanbowang, 2024-10-16:
I see, thank you for the advices.

closed 2024-10-16 by nathanbowang