pybricks/support #1903: [Feature] Set default main program

laurensvalk, 2024-10-28:
> My team is putting one file per mission model and one issue is that they often hit play from their mission model file instead of the main file that loads all of them.  Any possibility of adding a "default" Play program file?

_Originally posted by @davidyang in https://github.com/pybricks/support/discussions/1901_

laurensvalk, 2024-10-28:
Thanks @davidyang for suggesting this.

I've been wondering about this too some times. Use cases:

- Working on a multi-mission program with one main entry point in another file (OPs case)
- Working on a library file and having a main program to test it

This is also conventional in full editors, where you might do this with a launch config.

**Alternatives considered**
Work with ``__name__ == "__main__"`` to make this work, but this can get a bit messy with many files.

**Implementation idea 1**

Mark file as main on file tab with an extra icon there.

**Implementation idea 2**

Enter main file in somewhere under settings.

**Implementation idea 3 (my favorite)**
Making this completely persistent in local storage as above could be confusing when you open the editor later to work on something else.

So here's a variant that lets you mark the main file by right clicking on a tab and choosing "mark as starting point".

In fact, since we don't currently have any context dialog for tabs, maybe just double clicking to toggle this will work.

If having an "entry icon" is tricky in this layout framework, we could prefix the displayed name with `(main)`.

[image]

davidyang, 2024-10-28:
Hey @laurensvalk - thanks for getting back to me so quickly.  For how we use pybricks, the kids don't really switch into and out of projects so having it as a setting that is persisted in localStorage makes sense to me.

If it's just in the tab (or right clicking a tab) I would be concerned about discoverability.  This does seem like something that would be in settings if I'm looking for it.  (Usually when I'm looking at a code project I'm looking for a run script somewhere in the home directory as a setting).

bcfletcher, 2025-05-01:
I posted a similar request at #2175

abliss, 2025-12-09:
I have this same issue. Is there some (albeit hacky) way to do the `if __name__ == "__main__": main()` trick today using blocks?

derrickgw, 2025-12-10:
A great pythonic way to create a main project file would be to search for a __init__.py file.  If it exists, it becomes the main program.  From there it could point to a main module or just import the main program.  It could also have a different icon in the GUI to help people know it is a special project level file.

Later you could make it fancy and have special menu blocks that are only available in this project file.

derrickgw, 2025-12-11:
Right now there isn't any support for projects, so I would say supporting a single project would be a first step.  Users who don't need/want a project can continue with the existing behavior.  This could add support for a multi-file project without any UI changes.

Long term the GUI could be updated to have the notion of folders/projects so that users can have multiple projects.  Then add some visual indicator (bold, collapse/expand folders, etc) of which project is active based on the current editor tab.