pybricks/support #1905: [Bug]

BertLindeman, 2024-10-28:
**Describe the bug**
related to #1887
What is the problem?
The documented default value for imu.acceleration_threshold  is 250 mm/s2 
```
acceleration_threshold (Number, mm/s²)
The threshold for angular velocity.
The default value is 250 mm/s².
```

The installed default is 2500.
There might already be an issue or discussion where this it documented, I did not find it.

Seen on 
. 3.5.0 at github hash 6c633a8dd
. 3.6.0b1 at github hash 6e56bde4
. 3.6.0b2 at github hash beaaf231

**To reproduce**
Steps to reproduce the behavior:
1. run a similar program like the example below
2. see 2500 being reported as imu.acceleration_threshold 

**Expected behavior**
What did you expect to happen instead?
250 as default imu.acceleration_threshold 

**Screenshots**
none now

**example program**
```python

from pybricks.hubs import ThisHub
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop, Axis
from pybricks.tools import wait
from pybricks import version
print(version)

hub = ThisHub()

imu_setting = []
all_imu_settings = hub.imu.settings()
imu_settings_len = len(all_imu_settings)

for i in range(imu_settings_len):
    imu_setting.append(all_imu_settings[i])

imu_angular_velocity_threshold = all_imu_settings[0]
imu_angular_accel_threshold = all_imu_settings[1]
# imu_angle_correction = all_imu_settings[2] if imu_settings_len >2 else 0

print(f'imu thresholds velocity: {imu_angular_velocity_threshold:>8.2f} deg/s', end="")
print(f' accel: {imu_angular_accel_threshold:>8.2f} mm/s2')
# print(f' angle_correction {imu_angle_correction:8.2f} deg')

```

laurensvalk, 2024-10-29:
Please see https://github.com/pybricks/support/issues/1105. Thanks!

Suggestions for improved values are welcome. I'd like to change the averaging algorithm slightly so we can use lower defaults without causing problems in noisy environments.

closed 2024-10-29 by laurensvalk

BertLindeman, 2024-10-29:
Sorry I made the change the wrong way.
Was fun trying to make the PR though.
Thank you, Laurens