pybricks/support #1907: [Feature] [Feature] User IMU calibration implementation

laurensvalk, 2024-10-29:
The technical aspects of calibration are being discussed in https://github.com/pybricks/support/issues/943, this issue is more about the user routine: How can we make it easy for the user to calibrate their gyro? This is an alternative to https://github.com/pybricks/support/issues/1678 to be ready for 3D. It is a bit simpler too since you don't need an external reference.

This would have to be done just once and the results would be saved persistently on the hub. You'd have to do it again if you updated the firmware. This will calibrate the accelerometer and gyro all in one go.

The procedure is as follows, three times. It takes a bit of practice, but it is easy to do.
- Place the hub in an indicated starting position.
- Roll it towards you 4 times without lifting it up.

We could animate this visually in a Pybricks Code pane. This is roughly what the user would have to do:

https://github.com/user-attachments/assets/b118ad6b-5b84-4ee1-8175-d16cc2676f99

 (Video is cut off. This repeats one more time for the final axis.)

**Instructions**
<details>

<summary>Click to try it out**</summary>

- Install [this firmware](https://github.com/pybricks/pybricks-micropython/pull/270#issuecomment-2437391848).
- Run the following. Rough instructions are shown in Pybricks Code. (We'll clean that up later.)

```python
from pybricks.hubs import ThisHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop, Axis
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch, vector

hub = ThisHub()

def wait_for_stationary(side):
    while not hub.imu.stationary() or hub.imu.up() != side:
        wait(10)

up_sides = {
    Side.FRONT: (0, 0, 1),
    Side.BACK: (1, 0, -1),
    Side.LEFT: (2, 1, 1),
    Side.RIGHT: (3, 1, -1),
    Side.TOP: (4, 2, 1),
    Side.BOTTOM: (5, 2, -1),
}

gravity = [0] * 6
bias_count = 0
bias = vector(0, 0, 0)

_, _, _, old_scale, _, = hub.imu.settings()

def roll_over_axis(axis, new_side, first=False):

    global bias, bias_count

    if first:
        print("Roll it over towards you, without lifting the hub up!")
    else:
        print("Roll towards you again!")

    angle_start = hub.imu.rotation(axis)
    while hub.imu.up() != new_side or not hub.imu.stationary():
        
        x, y, z = hub.imu.orientation() * axis
        if abs(z) > 0.05:
            print(hub.imu.orientation() * axis)
            raise RuntimeError("Oops, lifted it!") 

        wait(100)

    rotation = abs(hub.imu.rotation(axis)-angle_start)
    if abs(rotation - 90) > 10:
        raise RuntimeError("Rotated too far or not far enough!")

    print("Nice work. Calibrating now...")
    hub.speaker.beep(1000)
    wait(10)

    rotation_start = vector(
        hub.imu.rotation(Axis.X),
        hub.imu.rotation(Axis.Y),
        hub.imu.rotation(Axis.Z)
    )

    COUNT = 1000
    acceleration = vector(0, 0, 0)
    
    for i in range(COUNT):
        acceleration += hub.imu.acceleration(calibrated=False)        
        bias += hub.imu.angular_velocity(calibrated=False)
        bias_count += 1
        wait(1)

    acceleration /= COUNT

    rotation_end = vector(
        hub.imu.rotation(Axis.X),
        hub.imu.rotation(Axis.Y),
        hub.imu.rotation(Axis.Z)
    )
    if abs(rotation_end - rotation_start) > 1:
        raise RuntimeError("Too much rotation while calibrating")

    side_index, axis_index, axis_direction = up_sides[new_side]

    # Store the gravity value for the current side being up.
    # We will visit each side twice. The second time the gravity value will
    # already be nonzero, so we can take the average between two to get
    # slightly better results while we're here anyway.
    acceleration_previous = gravity[side_index]
    acceleration_now = acceleration[axis_index]
    if abs(acceleration_previous) > 1:
        acceleration_now = (acceleration_now + acceleration_previous) / 2
    gravity[side_index] = acceleration_now

    # Todo, expose unscaled version in api?
    unadjusted_rotation = rotation * old_scale[axis_index] / 360

    hub.speaker.beep(500)

    return unadjusted_rotation

calibrate_x = """
We're going to calibrate X now!
- Put the hub on the table in front of you.
- top side (display) facing up
- right side (ports BDF) towards you.
"""

calibrate_y = """
We're going to calibrate Y now
- Put the hub on the table in front of you.
- top side (display) facing up
- back side (speaker) towards you.
"""

calibrate_z = """
We're going to calibrate Z now!
- Put the hub on the table in front of you.
- front side (USB port) facing up
- left side (ports ACE) towards you
"""

print(calibrate_x)
wait_for_stationary(Side.TOP)
hub.speaker.beep()

rotation_x = roll_over_axis(Axis.X, Side.LEFT, first=True)
rotation_x += roll_over_axis(Axis.X, Side.BOTTOM)
rotation_x += roll_over_axis(Axis.X, Side.RIGHT)
rotation_x += roll_over_axis(Axis.X, Side.TOP)

print(calibrate_y)
wait_for_stationary(Side.TOP)
hub.speaker.beep()

rotation_y = roll_over_axis(Axis.Y, Side.FRONT, first=True)
rotation_y += roll_over_axis(Axis.Y, Side.BOTTOM)
rotation_y += roll_over_axis(Axis.Y, Side.BACK)
rotation_y += roll_over_axis(Axis.Y, Side.TOP)

print(calibrate_z)
wait_for_stationary(Side.FRONT)
hub.speaker.beep()

rotation_z = roll_over_axis(Axis.Z, Side.RIGHT, first=True)
rotation_z += roll_over_axis(Axis.Z, Side.BACK)
rotation_z += roll_over_axis(Axis.Z, Side.LEFT)
rotation_z += roll_over_axis(Axis.Z, Side.FRONT)

print("old: ", hub.imu.settings())

print("applying")

print(bias[0] / bias_count, bias[1] / bias_count, bias[2] / bias_count)
print(rotation_x, rotation_y, rotation_z)
print(gravity)

hub.imu.settings(
    angular_velocity_bias = (bias[0] / bias_count, bias[1] / bias_count, bias[2] / bias_count),
    angular_velocity_scale = (rotation_x, rotation_y, rotation_z),
    acceleration_correction = gravity
)

print("new: ", hub.imu.settings())
```

</details>

laurensvalk, 2024-10-29:
To see the "before" and "after" situation, try running the following script before and after doing the calibration routine:

```
from pybricks.hubs import ThisHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop, Axis
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch, vector
from umath import acos, degrees, pi, sqrt

hub = ThisHub()

startx = hub.imu.rotation(Axis.X)
starty = hub.imu.rotation(Axis.Y)
startz = hub.imu.rotation(Axis.Z)

while True:

    if hub.buttons.pressed():

        startx = hub.imu.rotation(Axis.X)
        starty = hub.imu.rotation(Axis.Y)
        startz = hub.imu.rotation(Axis.Z)

        while hub.buttons.pressed():
            wait(10)

    x = hub.imu.rotation(Axis.X) - startx
    y = hub.imu.rotation(Axis.Y) - starty
    z = hub.imu.rotation(Axis.Z) - startz

    print(f"{x:10.3f}", f"{y:10.3f}", f"{z:10.3f}", f"{hub.imu.heading():10.3f}")

    wait(100)
```

You might find that before, a flat rotation around each of the axis is not a perfect 360-degree turn. After calibration, it should be. In one of my hubs, this corrects a y-axis rotation from 354 to 360, which is quite a win.

BertLindeman, 2024-10-29:
Before calibration: ~364.something and after 360.9. Great.

Was a bit tricky to see what side of the robot inventor has the speaker due to the lighting this time of day.
To remind me I added "(under the center button)"

Quite doable calibration scenario.

'naughty ON:' Curious how you will suggest to do the Technic hub 😜 'naughty OFF'

Even if a beam is added at the TechnicHub sides it wobbles on the small ridges on the sides. (not LEGO like)

Tried the Technichub anyway (ignore if I am out of line) see picture below.

All goes well until calibration of the Z side:
```
We're going to calibrate Z now!
- Put the hub on the table in front of you.
- front side (USB port) facing up
- left side (ports ACE) towards you

Roll it over towards you, without lifting the hub up!
Matrix([
    [   0.082],
    [  -0.995],
    [   0.053],
])
Traceback (most recent call last):
  File "tool_issue_1907_calibrate_3d.py", line 164, in <module>
  File "tool_issue_1907_calibrate_3d.py", line 47, in roll_over_axis
RuntimeError: Oops, lifted it! abs(z) 0.05343884
```
The z is just a small bit over 0.05. 

testhub:
[image]

Great step forwards, thank you!

Bert

BertLindeman, 2024-10-29:
Stubborn as I am, got a Technic hub calibrated.
From 367.069  and after calibration -360.862.
That ~~might be~~ is a matchwinner. 

I have set the z allowance from 0.05 to 0.06 and used this support for the technichub with 3L axle and "connectors number 1" with half pins in them:
[image]

laurensvalk, 2024-12-03:
The routine is now built into the latest firmware. I have updated the testing instructions. API updates are to follow...

BertLindeman, 2024-12-03:
Technic hub:
```
Pybricks MicroPython ci-build-3625-v3.6.0b2-24-g77eb7529 on 2024-12-03; LEGO Technic Hub with STM32L431RC
Type "help()" for more information.
>>> import _imu_calibrate 

Going to calibrate X now!
- Put the hub on the table in front of you.
- top side (display) facing up
- right side (ports BDF) towards you.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "_imu_calibrate.py", line 134, in <module>
  File "_imu_calibrate.py", line 125, in roll_hub
TypeError: beep() takes 1 positional arguments but 0 were given
>>> 
```

laurensvalk, 2024-12-03:
Fixed now. Firmware is being built... Stay tuned.

BertLindeman, 2024-12-03:
Technic hub at ```ci-build-3628-v3.6.0b2-25-g9e5cd4ca```:
```
Result:  (2.0, 2500.0, (-1.053093, 0.2765429, 0.01441465), (361.6532, 361.6852, 365.6783), (9984.508, -9751.642, 9935.678, -9673.925, 10030.06, -9721.781), 360.0)
```
(Primehub is in a model and the RI-hub still is in "LEGO-mode")

DrTom, 2024-12-07:
Just tried that with the nightly primehub-firmware-build-3641-git40cc0613.zip. The process starts but when I start rolling it claims that I lifted the hub:

```
W49-6-Sat_10:11:39 Going to calibrate X now!
W49-6-Sat_10:11:39 - Put the hub on the table in front of you.
W49-6-Sat_10:11:39 - top side (display) facing up
W49-6-Sat_10:11:39 - right side (ports BDF) towards you.
W49-6-Sat_10:11:39 
W49-6-Sat_10:11:39 Roll it towards you, without lifting the hub up!
W49-6-Sat_10:11:42 Matrix([
W49-6-Sat_10:11:42     [   0.892],
W49-6-Sat_10:11:42     [  -0.444],
W49-6-Sat_10:11:42     [   0.080],
W49-6-Sat_10:11:42 ])
W49-6-Sat_10:11:42 Traceback (most recent call last):
W49-6-Sat_10:11:42   File "test_imu_calibration.py", line 137, in <module>
W49-6-Sat_10:11:42   File "test_imu_calibration.py", line 132, in roll_hub
W49-6-Sat_10:11:42   File "test_imu_calibration.py", line 53, in roll_over_axis
W49-6-Sat_10:11:42 RuntimeError: Lifted it!
```

roykrikke, 2024-12-14:
I'm having issues. Could someone help me please?

I'm running it from the https://beta.pybricks.com/ with the beta firmware.

```
Pybricks MicroPython ci-release-81-v3.6.0b2 on 2024-10-15; SPIKE Prime Hub with STM32F413VG
Type "help()" for more information.
>>> import _imu_calibrate
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named '_imu_calibrate'
>>> 
```

laurensvalk, 2024-12-14:
Thanks for reporting! We should be able to update the test files in a few says.

roykrikke, 2024-12-15:
> Thanks for reporting! We should be able to update the test files in a few days.

I studied the code a bit, I can't figure out the error right now. Possibly lack of knowledge in python. Anyway... I'm using the pybrciks code online editor now. Is there a handy way that I can help find the error? E.g. IDE with step through debugging option?

laurensvalk, 2024-12-16:
It seems that the calibration passes but it does not succeed in saving it. It does some value checks to avoid saving bad values. For example, it will reject gyro corrections of more than 10 degrees as this might be a user error. But maybe on some hubs the errors are truly 11 degrees, for example.

So could you try running the following adapted script? You can just from it from the online editor. It is the same as before but now it will print the values you found before trying to save it. It should give the same error, but then we have an indication as to why it might be. And then we can relax the tolerances.

<details>

<summary>Click to see adapted script.</summary>

```python
from pybricks.hubs import ThisHub
from pybricks.parameters import Side, Axis
from pybricks.tools import wait, vector

hub = ThisHub()

def beep(freq):
    try:
        hub.speaker.beep(freq, 100)
    except AttributeError:
        # Technic hub does not have a speaker.
        pass
    wait(10)

def wait_for_stationary(side):
    while not hub.imu.stationary() or hub.imu.up() != side:
        wait(10)

up_sides = {
    Side.FRONT: (0, 0),
    Side.BACK: (1, 0),
    Side.LEFT: (2, 1),
    Side.RIGHT: (3, 1),
    Side.TOP: (4, 2),
    Side.BOTTOM: (5, 2),
}

gravity = [0] * 6
bias = vector(0, 0, 0)

STATIONARY_COUNT = 1000

def roll_over_axis(axis, new_side):

    global bias, bias_count

    print("Roll it towards you, without lifting the hub up!")

    angle_start = hub.imu.rotation(axis, calibrated=False)
    while hub.imu.up() != new_side or not hub.imu.stationary():

        _, _, z = hub.imu.orientation() * axis
        if abs(z) > 0.07:
            print(hub.imu.orientation() * axis)
            raise RuntimeError("Lifted it!")
        wait(100)

    uncalibrated_90_deg_rotation = abs(hub.imu.rotation(axis, calibrated=False) - angle_start)
    if abs(uncalibrated_90_deg_rotation - 90) > 10:
        raise RuntimeError("Not 90 deg!")

    print("Calibrating...")
    beep(1000)

    rotation_start = vector(
        hub.imu.rotation(Axis.X, calibrated=False),
        hub.imu.rotation(Axis.Y, calibrated=False),
        hub.imu.rotation(Axis.Z, calibrated=False),
    )

    acceleration = vector(0, 0, 0)

    for i in range(STATIONARY_COUNT):
        acceleration += hub.imu.acceleration(calibrated=False)
        bias += hub.imu.angular_velocity(calibrated=False)
        wait(1)

    acceleration /= STATIONARY_COUNT

    rotation_end = vector(
        hub.imu.rotation(Axis.X, calibrated=False),
        hub.imu.rotation(Axis.Y, calibrated=False),
        hub.imu.rotation(Axis.Z, calibrated=False),
    )
    if abs(rotation_end - rotation_start) > 1:
        raise RuntimeError("Moved it!")

    side_index, axis_index = up_sides[new_side]

    # Store the gravity value for the current side being up.
    # We will visit each side several times. We'll divide by the number
    # of visits later.
    gravity[side_index] += acceleration[axis_index]

    beep(500)

    return uncalibrated_90_deg_rotation

calibrate_x = """
Going to calibrate X now!
- Put the hub on the table in front of you.
- top side (display) facing up
- right side (ports BDF) towards you.
"""

calibrate_y = """
Going to calibrate Y now
- Put the hub on the table in front of you.
- top side (display) facing up
- back side (speaker) towards you.
"""

calibrate_z = """
Going to calibrate Z now!
- Put the hub on the table in front of you.
- front side (USB port) facing up
- left side (ports ACE) towards you
"""

REPEAT = 2

# For each 3-axis run, we will visit each side twice.
SIDE_COUNT = REPEAT * 2

def roll_hub(axis, message, start_side, sides):
    print(message)
    wait_for_stationary(start_side)
    beep(500)
    rotation = 0
    for _ in range(REPEAT):
        for side in sides:
            rotation += roll_over_axis(axis, side)
    return rotation / REPEAT

rotation_x = roll_hub(
    Axis.X, calibrate_x, Side.TOP, [Side.LEFT, Side.BOTTOM, Side.RIGHT, Side.TOP]
)
rotation_y = roll_hub(
    Axis.Y, calibrate_y, Side.TOP, [Side.FRONT, Side.BOTTOM, Side.BACK, Side.TOP]
)
rotation_z = roll_hub(
    Axis.Z, calibrate_z, Side.FRONT, [Side.RIGHT, Side.BACK, Side.LEFT, Side.FRONT]
)

print("angular_velocity_bias=", tuple(bias / SIDE_COUNT / STATIONARY_COUNT / 6))
print("angular_velocity_scale=", (rotation_x, rotation_y, rotation_z))
print("acceleration_correction=", [g / SIDE_COUNT for g in gravity])

hub.imu.settings(
    angular_velocity_bias=tuple(bias / SIDE_COUNT / STATIONARY_COUNT / 6),
    angular_velocity_scale=(rotation_x, rotation_y, rotation_z),
    acceleration_correction=[g / SIDE_COUNT for g in gravity],
)

print("Result: ", hub.imu.settings())
```

</details>

roykrikke, 2024-12-20:
See below, I cut out the obvious.
```
Going to calibrate X now!
- Put the hub on the table in front of you.
- top side (display) facing up
- right side (ports BDF) towards you.

Roll it towards you, without lifting the hub up!
Calibrating...

Going to calibrate Y now
- Put the hub on the table in front of you.
- top side (display) facing up
- back side (speaker) towards you.

Roll it towards you, without lifting the hub up!
Calibrating...

Going to calibrate Z now!
- Put the hub on the table in front of you.
- front side (USB port) facing up
- left side (ports ACE) towards you

Roll it towards you, without lifting the hub up!
Calibrating...

angular_velocity_bias= (0.07558279, -3.181849, 1.326638)
angular_velocity_scale= (366.9297, 349.4564, 363.9098)
acceleration_correction= [9898.266, -9692.415, 9925.33, -9830.211, 9589.978, -10110.9]
Traceback (most recent call last):
  File "test.py", line 150, in <module>
ValueError: Invalid argument
```

roykrikke, 2024-12-20:
> Thank you, this is very helpful!
> 
> Your hub is reporting the following amount of degrees for each full x, y, z rotation respectively:
> 
> ```
> 366.9297, 349.4564, 363.9098
> ```
> 
> This is further off the ideal 360 value than I have seen in any of our hubs. We had set the limits at 350-370, so this explains the issue: yours is just outside it.
> 
> We will increase the tolerances since this is clearly necessary, so that calibration will pass on your hub too.
> 
> When we do that and you complete this procedure again, you should get a massive improvement for measuring rotation around the Y axis for your hub!

Thanks for explaining this. Could you give an indication when you have new firmware as a beta? Furthermore I am curious when this feature will officially come as a release.

I started reading the C code and I played with:
print("Result: ", hub.imu.settings()) before the calibration.
Result is:
`Result:  (2.0, 2500.0, (0.07558279, -3.181849, 1.314846), (366.9297, 360.0, 360.0), (9806.65, -9806.65, 9806.65, -9806.65, 9806.65, -9806.65), 360.0)`

laurensvalk, 2024-12-22:
Just increased the tolerances from 10 to 15 degrees, which should cover your hub. You can follow the original instructions to get the new, latest build.

Thank you for trying this and reporting this. Now we could fix the tolerances before releasing it!

roykrikke, 2024-12-23:
I have seen a mega improvement. THANKS!

Before:
`(2.0, 2500.0, (0.1462465, -3.244734,  1.333473), (360.0,    360.0,    360.0),   (9806.65,  -9806.65, 9806.65, -9806.65,  9806.65,  -9806.65),  360.0)`
After
`(2.0, 2500.0, (0.08115859, -3.275208, 1.323259), (367.2673, 349.2709, 364.027), (9900.773, -9687.63, 9936.37, -9823.288, 9620.516, -10081.91), 360.0)`

Simple test
```
     0.000     -0.000     -0.000      0.000
   -90.253     -1.432     -8.037      8.037
  -180.402      0.514     -7.243      7.243
  -270.290      0.985     -5.475      5.475
  -360.067     -0.480     -4.675      4.675
```

Would It help if I would share the same measurement results as on https://github.com/pybricks/support/issues/1962#issuecomment-2516984193

MrGibbage, 2025-02-17:
We haven't used the IMU calibration yet, but I want to show it to the kids this summer. But first I want to run some tests so I can see what to expect. I understand that the calibration settings survive a reboot, so what I'd like to know is there a way to delete the settings once it has been calibrated? I want the kids to calibrate "uncalibrated" hubs so they can see the difference.

Also, how would I calibrate the hub when it is installed in a robot? Our robot is not very flat on top.
https://github.com/FLL-Team-24277/FLL-Fall-2024-Submerged/blob/main/help/images/Robot_summer_2023_CompetitionBot.png

We could take that "L" piece off the top (it's just a safety to keep us from accidentally hitting the button) and just hold it flat when it is time to go top-down, but we'd have to continuously hold it while doing that step. 

I'm going to look through the _imu_calibrate code, but is there anything you can say about what the code actually does and how it is used, say, when I use a drive() command after calibration?

laurensvalk, 2025-02-17:
Calibration is needed only once, so hopefully taking it out of your robot is not an issue.

However, there will also be a way to adjust the gyro value without doing that. I am currently working on the documentation for all of this 🙂 

>  is there a way to delete the settings once it has been calibrated?

This will erase settings and your program:

```
hub.system.reset_storage()
```

MrGibbage, 2025-02-17:
Unfortunately, getting the hub out of the robot is quite an ordeal. Serious Lego-finger! But if that is what we have to do, then we will do it.

roykrikke, 2025-03-02:
Hello @laurensvalk is this correct `import _imu_calibrate` and then see the results with `import _imu_calibrate` 

```
>>> print(hub.imu.settings())
(2.0, 2500.0, (0.08999829, -2.871182, 1.170795), (367.9688, 349.4978, 363.1353), (9888.416, -9705.651, 9926.816, -9822.398, 9580.736, -10112.13), 360.0)
```
Then I do a `hub.system.reset_storage()`

Still I would like the results from before.

[image]

To put 367.9688 in heading_correction in the program?

laurensvalk, 2025-03-02:
If you successfully ran the calibration routine like you did, you shouldn’t
have to manually set the heading correction. The heading should already be
very consistently close to 360 now.

The manual heading correction can be used if you don’t want to (or can’t)
run the calibration routine. They can be used together technically but this
should rarely be needed.

zoligyenge, 2025-03-07:
> Calibration is needed only once, so hopefully taking it out of your robot is not an issue.

Does this mean I do not need to re-calibrate after each firmware upgrade?

roykrikke, 2025-03-16:
> If you successfully ran the calibration routine like you did, you shouldn’t
> have to manually set the heading correction. The heading should already be
> very consistently close to 360 now.
>
> The manual [...]

I think I didn’t explain my question before not clearly enough. What I meant is exactly what @zoligyenge is asking as well. I believe I read somewhere that the calibration data is lost after a firmware update.

roykrikke, 2025-03-30:
@laurensvalk see my previous message. Is the calibration data indeed lost after a firmware update?

laurensvalk, 2025-03-30:
Yes, but technically speaking it is only lost if you update to a _different_ version.

If you reinstall the same version, it can't tell the difference so it keeps all user data.

roykrikke, 2025-08-06:
> > Quick question — I'm currently on holiday so I can't test it myself, but I was wondering: is this feature now part of the stable release by default?
> 
> Yes, 3D heading and calibration were implemented in https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.0b3 and therfore part of subsequent versions such as the current stable release 3.6.1. Calibration data is still lost after firmware update, so you need to back your settings up and reinstall after the update. I believe this is intentional, because people could choose to install custom versions of the firmware with different memory layout e.g. to have more space for userprogram and drop ununsed pybricks build-in libraries.

Thanks for sharing, is of great help!

Another question: I've noticed that features in the repo keep their status open. I'm used to them being closed. Is there a house rule for this in the Pybrick repos?

davidyang, 2026-09-20:
@laurensvalk you are a genius - this was huge for setting up a new robot!  I'm wondering what else I don't know about in terms of calibration (can motor encoders also be calibrated?)

This is amazing, last season we had two motors that were producing different outputs and didn't really know how to fix it.