pybricks/support #1915: [Bug] pybricksdev HID does not work in all environments.

laurensvalk, 2024-11-04:
**Describe the bug**

```
 pybricksdev flash ev3/build/firmware.zip
```

only seems to work in some environments but not in others.

**To reproduce**

The actual flashing code seems irrelevant. Just testing on a local REPL will do.

In my `pybricksdev` development folder I can do:
```
(.venv) laurens@laurens-ubuntu:~/git-pybricks/pybricksdev$ python
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> dev = hid.device()
>>> dev.open(1684, 6)
>>> dev.close()
```

Anywhere else on the same system with the same privileges I get:

```
pybricks-micropython-py3.10laurens@laurens-ubuntu:~/git-pybricks/pybricks-micropython$ python
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> dev = hid.device()
>>> dev.open(1684, 6)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hidraw.pyx", line 143, in hid.device.open
OSError: open failed
>>> 
```

And this is after bumping the pybricksdev dependency there.

I can't think of any differences between these environments. Since this is the same system, the `udev` rules seem fine.

**Expected behavior**
Work the same everywhere on the same system.

dlech, 2024-11-04:
Just because pybricksdev was updated, it doesn't mean that other dependencies were updated. The `.pyx` file extension tells me that it has something to do with Cython, like it didn't get compiled correctly on install.

laurensvalk, 2024-11-05:
Thank you. ~~The offending package seems to be `hidapi` at `0.14.0.post"`. My `pybricksdev` development folder was still on `0.14.0`, which works. Forcing it to `=0.14.0` seems to work around the issue.~~

laurensvalk, 2025-01-20:
It looks like this has been fixed upstream. Our requirements already pull this in as `hidapi = "^0.14.0"` so reinstalling `pybricksdev` should do the trick:

```
Updating hidapi (0.14.0.post2 -> 0.14.0.post4)
```

laurensvalk, 2025-01-20:
Fixed by https://github.com/pybricks/pybricksdev/commit/d7869fb41b4cc99e7753f000cdc683cbdafbde2d