pybricks/support #1918: [Bug]

Sooutherncharm, 2024-11-05:
**Describe the bug**
What is the problem?

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

Wurzelteppich, 2024-11-19:
Describe the bug
An unexpected error occurred when trying to connect via Bluetooth in Pybricks.

To reproduce
Steps to reproduce the behavior
[image]
[image]

Go to code.pybricks.com.
Click on the Bluetooth icon (top left corner).
Attempt to establish a Bluetooth connection.
Observe the error message.
Expected behavior
The Bluetooth connection should have been successfully established, and I should have been able to proceed with coding on the Pybricks interface.

Screenshots
Attached are two screenshots showing the issue:

The first screenshot shows the Bluetooth connection attempt in Pybricks.
The second screenshot displays the error message that occurred after the attempt.
Screenshot 1:
Screenshot 2:

laurensvalk, 2024-12-03:
Did this issue persist, or did it go away when you tried again?

dlech, 2025-02-01:
Closing due to not enough info.

closed 2025-02-01 by dlech