pybricks/support #1962: [Feature] 3D-based heading for driving. Feedback wanted!

laurensvalk, 2024-12-04:
**What did we do?**
Until now, when you did `hub.imu.heading()` or `drivebase.use_gyro(True)`, it would only register "flat turns", like driving on a table. But if you lifted the hub or steered across a ramp, you would lose the angle.

We updated the algorithms to take full motion into account. In the future, `hub.imu.heading()` will do this by default.

**Gradual introduction**
To make sure nobody experiences sudden changes in the middle of the FLL season, we're introducing this gradually. For now, `hub.imu.heading()` will still only register flat turns by default.

**How can you try it?**
But if you install the [latest nightly firmware](https://pybricks.com/learn/getting-started/install-pybricks/#trying-the-nightly-version-optional), you can now do:

```python
# Old style heading. 
heading1 = hub.imu.heading()

# New style heading. This will become the default if everyone is happy.
heading3 = hub.imu.heading('3D')
```

It will work even better if you run [the new calibration first](https://github.com/pybricks/support/issues/1907). Just once. The results are saved on the hub until you change firmware again.

**Copy pastable example**

```python
from pybricks.hubs import PrimeHub
from pybricks.tools import wait

# Optional: For more accuracy, calibrate once.
# Here's how: https://github.com/pybricks/support/issues/1907

# You can also use Technic/Essential/Inventor Hub 
hub = PrimeHub() 

while True:

    # Old style heading. 
    heading1 = hub.imu.heading()

    # New style heading. This will become the default if everyone is happy.
    heading3 = hub.imu.heading('3D')

    # It should be approximately the same as long as you stay flat. But the new
    # version should be much better if you lift it up and put it back.

    print(f"Old: {heading1:7.3f}    New: {heading3:7.3f}", )

    wait(100)

```

**Can I use it with drive bases too?**

Yes! As above, "flat turns" will remain default for now. But you can now do:
- Same as before: `drivebase.use_gyro(False)`
- Same as before: `drivebase.use_gyro(True)`. But instead you can now do: `drivebase.use_gyro('3D')` to try out the future method.

Now it should work better on ramps or when it drives over bumps and obstacles.

**Can I use it with block coding?**

In the future, this will be the default so no action is needed, whether you use Python or blocks. If you want to try it early with blocks, please let me know and we'll put an example together. Thank you!

**What are the limitations?**
It is still just a gyro without a compass for corrections. If you lift it from a table and make lots of motion and put it back, it's going to be off by a few degrees. But still much less than the "flat" case.

**What do we want to know from you?**
Before we make this the default for everyone, we want to make sure that `hub.imu.heading('3D')` is **at least as accurate** as `hub.imu.heading()` for everyday use like on an FLL table with mostly flat turns.

We'd love to know if you want to try this in an FLL or WRO practice session. 

If you do the new calibration, please let us know how it went. Just that alone should give you an accuracy bonus.

Thank you for your time!

If you want to do this as a science project with your students, we're happy to help, and answer "interview questions"!

laurensvalk, 2024-12-04:
Thanks @afarago , @DrTom, @ggramlich, MonongahelaCryptidCooperative, @TheWendyPower, @BertLindeman for providing feedback about using the gyro in the past year. As you can see, it has now gone to the next stage with lots of new things to try. Thank you for your consideration!

BertLindeman, 2024-12-04:
Impressive Laurens!

Added 
```python
hub.imu.reset_heading(0)
```

and stationary on the print:

```python
    print(f"Old: {heading1:7.3f}    New: {heading3:7.3f} {hub.imu.stationary()}")
```

Ran on an **not** calibrated TechnicHub.
Put it flat against the chair, turn it up right and rotate 90 degrees and back and flat again.

End of the measurements:
```
Old:   0.003    New:   0.002 False
Old:   0.020    New:   0.020 False
Old:   0.166    New:   0.163 False
Old:   1.150    New:   1.154 False
Old:   2.980    New:   2.977 False
Old:   3.343    New:   3.352 False
Old:   1.234    New:   1.256 False
Old:   0.238    New:   0.229 False
Old:  -0.375    New:  -0.365 False
Old:  -1.437    New:  -1.435 False
Old:  -2.547    New:  -2.551 False
Old:  -3.679    New:  -3.663 False
Old:  -4.705    New:  -4.696 False
Old:  -5.155    New:  -5.243 False
Old:  -4.569    New:  -4.768 False
Old:  -1.747    New:  -1.878 False
Old:  -1.214    New:  -1.356 False
Old:  -0.631    New:  -0.778 False
Old:  -0.643    New:  -0.976 False
Old:  -0.518    New:  -1.070 False
Old:  -0.476    New:  -1.365 False
Old:  -0.457    New:  -1.546 False
Old:  -0.459    New:  -1.711 False
Old:  -0.453    New:  -2.403 False
Old:  -0.520    New: -14.731 False
Old:  -0.501    New: -11.165 False
Old:  -0.483    New:  -7.026 False
Old:  -0.542    New:  -7.612 False
Old:  -0.600    New:  -9.483 False
Old:  -0.855    New: -23.168 False
Old:  -0.981    New: -28.151 False
Old:  -1.120    New: -28.924 False
Old:  -1.160    New: -29.621 False
Old:  -1.178    New: -31.567 False
Old:  -1.170    New: -25.196 False
Old:  -1.218    New:   3.472 False
Old:  -1.373    New:  59.658 False
Old:  -1.262    New:  74.127 False
Old:  -1.124    New:  84.129 False
Old:  -0.935    New:  90.710 False
Old:  -0.738    New:  91.884 False
Old:  -0.660    New:  86.631 False
Old:  -0.561    New:  79.924 False
Old:  -0.614    New:  25.053 False
Old:  -0.654    New:  15.859 False
Old:  -1.101    New:  12.823 False
Old:  -4.446    New:   7.267 False
Old: -11.121    New:  -0.569 False
Old: -11.378    New:  -1.032 False
Old: -11.926    New:  -1.908 False
Old: -11.253    New:  -1.260 False
Old: -14.396    New:  -4.401 False
Old: -17.607    New:  -7.592 False
Old: -16.447    New:  -6.475 False
Old: -16.207    New:  -6.257 False
Old: -15.963    New:  -6.009 False
Old: -14.827    New:  -4.857 False
Old: -14.328    New:  -4.370 False
Old: -14.086    New:  -4.100 False
Old: -16.467    New:  -6.438 False
Old: -16.829    New:  -6.890 False
Old: -16.673    New:  -6.716 False
Old: -15.912    New:  -5.949 False
Old: -15.517    New:  -5.583 False
Old:  -8.704    New:   1.259 False
Old:  -8.700    New:   1.260 False
Old:  -8.668    New:   1.291 False
Old:  -8.689    New:   1.269 False
Old:  -8.706    New:   1.251 False
Old:  -8.725    New:   1.229 False
Old:  -8.621    New:   1.335 False
Old:  -8.400    New:   1.559 False
Old:  -8.438    New:   1.520 False
Old:  -8.433    New:   1.523 False
Old:  -8.431    New:   1.523 False
Old:  -8.430    New:   1.524 False
Old:  -8.429    New:   1.524 False
Old:  -8.429    New:   1.524 False
Old:  -8.428    New:   1.524 False
Old:  -8.427    New:   1.525 False
Old:  -8.432    New:   1.520 False
Old:  -8.432    New:   1.519 False
Old:  -8.429    New:   1.521 True
Old:  -8.425    New:   1.526 True
Old:  -8.420    New:   1.530 True
Old:  -8.412    New:   1.538 True
Old:  -8.410    New:   1.539 True
Old:  -8.410    New:   1.540 True
Old:  -8.408    New:   1.542 True
Old:  -8.404    New:   1.545 True
Old:  -8.404    New:   1.545 True
Old:  -8.401    New:   1.548 True
Old:  -8.399    New:   1.550 True
Old:  -8.398    New:   1.551 True
Old:  -8.396    New:   1.552 True
Old:  -8.395    New:   1.554 True
Old:  -8.394    New:   1.554 True
Old:  -8.393    New:   1.556 True
Old:  -8.392    New:   1.556 True
Old:  -8.390    New:   1.559 True
Old:  -8.389    New:   1.559 True
Old:  -8.387    New:   1.561 True
Old:  -8.387    New:   1.561 True
Old:  -8.386    New:   1.562 True
Old:  -8.385    New:   1.564 True
Old:  -8.383    New:   1.566 True

```

roykrikke, 2025-08-06:
> I asked a question in: [#1907 (comment)](https://github.com/pybricks/support/issues/1907#issuecomment-3154279338)
> 
> The same question is applicable to this feature also. I'm curious about the reaction

My question was answered in: https://github.com/pybricks/support/issues/1907#issuecomment-3156109994

roykrikke, 2025-08-06:
> > I asked a question in: [#1907 (comment)](https://github.com/pybricks/support/issues/1907#issuecomment-3154279338)
> > The same question is applicable to this feature also. I'm curious about the reaction
> > Another question: I've noticed that features in the repo keep their status open. I'm used to them being closed. Is there a house rule for this in the Pybrick repos?
> 
> This issue is a request for comments. So far, one person has tried it and given feedback.
> 
> We usually aim for a bit more feedback before making a big change to an existing feature. It should be an improvement, but that is what the feedback phase is for. Once it is merged (or in this case, made default), the issue can be closed.
> 
> If you've tried it, what do you think?

@Debenben wrote:
"Yes, 3D heading and calibration were implemented in https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.0b3 and therfore part of subsequent versions such as the current stable release 3.6.1."
Link: https://github.com/pybricks/support/issues/1907#issuecomment-3156109994

I did not try yet, see my feedback/question in the link above.

1: I'm a bit surprised that @Debenben reaction contradicts yours @laurensvalk. It's not a big deal, but I thought it was already in an official release.
2: I will try this when am back from my vacation and share feedback.