pybricks/support #1964: [Bug]  Hub doesn't blink when trying to flash it

redg25, 2024-12-04:
**Describe the bug**
What is the problem? I got a Porsche GT4 Lego technic for my son. It comes with a Technic Hub. We have been able to connect it to the phone app but when trying to use it with Pybricks, we can't seem to be able to flash it. I follow the steps, I make sure it is charged and off but then I press the button and keep holding it, the light never comes, doesn't turn on at all. If I just press on once then it turns on but this is not what is required for flashing it. If I still click on install, then nothing happens it just scanning trying to pair something but it never comes. 
I have tried on Windows, Linux and Android, and edge, Chrome and Firefox. Always the same result, which is nothing. The only difference is that on Linux and Android, the device's bluetooth is able to pair with the Hub.

**To reproduce**
Steps to reproduce the behavior:
1. Go to Pybricks.code
2. Try to install firmware following steps
3. In the press hub and wait for light purple light steps, nothing happens, no light.

**Expected behavior**
What did you expect to happen instead? I expected the light to turn on as explain in the instructions

**Screenshots**
None

dlech, 2024-12-04:
The hub that comes with the Porche is a "Technic Move hub" which is different from a "Technic hub". Unfortunately, this new "Technic Move hub" uses a different bootloader and requires signed firmware, so we won't be able to run Pybricks on it.

You can see pictures of the supported hubs in the docs: https://docs.pybricks.com/en/stable/

redg25, 2024-12-04:
Thanks for your answer, maybe one day. Great job with the pybricks project.

Yahoo Mail: Search, organise, conquer

closed 2024-12-04 by dlech