pybricks/support #1981: [Bug] If program have input number with "0" start, import seems will no downlaod to brick.

a10036gt, 2024-12-19:
**Describe the bug**
If program have input number with "0" start (e.g. print(50) -> print(050)), import seems will no downlaod to brick. Some time just wrongly type extra 0 in front of the original input number, but if there is no import, program still work.

**To reproduce**
Steps to reproduce the behavior:
1. Create a Main program
2. Create a Lib program for import (from IssueTest_Lib import test, import IssueTest_Lib as lib, from IssueTest_Lib import *)
3. In Main Porgram import Lib, and do some simple staff like wait and print
4. Change number input from nn to 0nn.

**Expected behavior**
If no import, seems program works fine even number input is "0nn"

**Screenshots**

https://github.com/user-attachments/assets/7eb254ec-b470-4725-b2e6-ea08c02786de

BertLindeman, 2024-12-19:
Confirmed on ```('technichub', '3.6.0b2', 'ci-build-3639-v3.6.0b2-33-g8137ff5e on 2024-12-05')```

with program magweg.py:
```python
import magweg
print(0300)
```
reports:
```
Traceback (most recent call last):
  File "magweg.py", line 2, in <module>
ImportError: no module named 'magweg'
```

Without the leading zero:
```python
import magweg
print(10300)
```

reports (twice due to the import):
```
10300
10300
```

Problem in pre-parse?

laurensvalk, 2024-12-19:
It sounds like this might be a similar issue to:

> Apparently, this is caused by the Python parser that we are using to find resolve imports. https://github.com/pybricks/python-program-analysis

_Originally posted by @dlech in https://github.com/pybricks/support/issues/1804#issuecomment-2336868812_

BertLindeman, 2024-12-19:
@a10036gt  in python 3 the leading zero poses a syntax error:
```    print(0300)
          ^
SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers
```

If you really want to use an octal value it should be ```print(0o300)```

a10036gt, 2024-12-19:
@BertLindeman Yes I know it's a syntax error, but we find this big because while students tuning a huge program they wrongly have a leading zero (like adjust parameters but typos), but the error throw out is something not import, so they use a bit of time to find out what happened.

closed 2026-09-21 by laurensvalk