pybricks/support #2014: [Bug] Tried the Advanced adding zip file. Did not work

onthuhlist, 2025-01-13:
**Describe the bug**
What is the problem?
pybricks webapp said that firmware-base.bin was missing, but it is indeed in the .zip file.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?
I expected to be able to load the firmware.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

laurensvalk, 2025-01-13:
Which firmware file have you tried, and where did you find it?

This might help us reproduce the issue.

onthuhlist, 2025-01-14:
I followed the info from a github to make my own zip file. But the author
also had pre-made zip files and one of those worked. So nevermind.

closed 2025-01-14 by laurensvalk