pybricks/support #2024: [Bug] Compiling with GCC >= 13 results in broken firmware

dlech, 2025-01-19:
**Describe the bug**

Compiling the firmware with GCC 13 or 14 causes the stop button to stop working.

**To reproduce**

1. Build firmware with GCC 13 or 14.
2. Run a simple program that runs forever, e.g. `while True: pass`.
3. Press the stop button on the hub (or in Pybricks Code). Nothing happens, light matrix keeps on spinning, hub continues to send BLE status messages with user program running flag set.
4. Press the stop button again. Light matrix freezes. Hub resets after a second or so.

**Expected behavior**
Stop button should stop the program just as if the firmware was built with GCC 12.

dlech, 2025-01-19:
I tried removing `-flto` temporarily since that is a common source of breakage, but it didn't make a difference.

dlech, 2025-01-20:
I'm not able to reproduce this issue today, so closing.

closed 2025-01-20 by dlech

laurensvalk, 2025-01-28:
Could this have anything to do with it?

There is some special handling for `__GNUC__ >= 13`

```
void mp_stack_ctrl_init(void) {
    #if __GNUC__ >= 13
    #pragma GCC diagnostic push
    #pragma GCC diagnostic ignored "-Wdangling-pointer"
    #endif
    volatile int stack_dummy;
    MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
    #if __GNUC__ >= 13
    #pragma GCC diagnostic pop
    #endif
}
```

dlech, 2025-01-28:
No, this was completely in my imagination. There is another bug in our code that was causing the problem that I haven't filed yet.