pybricks/support #2038: [Bug] make is not installed and can't(?) be installed

petusky, 2025-01-29:
**Describe the bug**
I am unable to install a driver for a USB wifi adapter because make is not installed.  (Note that I am connected to WIFI via USB to my computer.). 

@volansky describes the need to install make in this post from 2016: https://github.com/ev3dev/ev3dev/issues/703 

However, I'm unable to install make (using a command like sudo apt-get update).  I get the following:

The repository 'http://security.debian.org stretch/updates Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: The repository 'http://httpredir.debian.org/debian stretch Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Is there a workaround for this security issue? Or another way to install make on the EV3?

(Mine is running  GNU bash, version 4.4.12(1)-release (arm-unknown-linux-gnueabi) )

**To reproduce**
Steps to reproduce the behavior:
Connect to the EV3 using SSH.  Issue "sudo apt-get update".

**Expected behavior**
I thought that would install make for me.

**Screenshots**
Other efforts to install make fail:

[image]

Here's the screen shot of the warning:

[image]

dlech, 2025-01-30:
Known issue. Please see step 6.1 in the getting started guide on how to fix this.

https://www.ev3dev.org/docs/getting-started/

petusky, 2025-01-30:
Thank you. That feedback was both timely and helpful and helped move me along.  However, I'm still not all the way there. The bottom line is that I'm missing a build directory needed for make.  I've tried updating Linux headers but that didn't fix the problem.  

**Here's what I've done:**

1) Updated /etc/apt/sources.list, per the instructions  in Step 6.1 referenced by @dlech.

2) Update the kernel:
**sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade**

3) Installed build-essentials to install make:
**sudo apt-get install build-essential**

4) Downloaded the WIFI adapter driver and copied to EV3 using scp.  From terminal on the mac (not the SSH session):
**scp ./EW-7811Un_V2_Linux_Driver_1.0.1.3.zip [robot@ev3dev.local](mailto:robot@ev3dev.local):/home/robot**

5) Unzipped on SSH:
**unzip EW-7811Un_V2_Linux_Driver_1.0.1.3**

6) Per instructions for the Edimax wifi adapter instructions on the [amazon page](https://www.amazon.com/Edimax-802-11n-Adapter-Smallest-Plug-n-Play-dp-B08F2ZNC6J/dp/B08F2ZNC6J/ref=dp_ob_title_ce), I updated headers:
**sudo apt install linux-headers-$(uname -r)**

7) I installed bc, after make warned me that it was not found:
**sudo apt install bc**

8) I navigated to the folder with the driver (/home/robot/EW-7811Un_V2_Linux_Driver_1.0.1.3) and invoke make:
**make**

But I get an error that the build directory is missing (/lib/modules/4.14.117-ev3dev-2.3.5-ev3/build). I've navigated there and indeed it is missing:

**robot@ev3dev:~/EW-7811Un_V2_Linux_Driver_1.0.1.3$ make
make ARCH=armv5tejl CROSS_COMPILE= -C /lib/modules/4.14.117-ev3dev-2.3.5-ev3/build M=/home/robot/EW-7811Un_V2_Linux_Driver_1.0.1.3  modules
make[1]: *** /lib/modules/4.14.117-ev3dev-2.3.5-ev3/build: No such file or directory.  Stop.
Makefile:2461: recipe for target 'modules' failed
make: *** [modules] Error 2**

I've researched this missing directory problem, but all the posted solutions I found tell me to deal with steps I've already tried above (such as updating Linux headers.). So at this point I'm stuck.  Help, please!

dlech, 2025-01-31:
> `-C /lib/modules/4.14.117-ev3dev-2.3.5-ev3/build`

Hmm... I don't think the files from that directory ever got packaged with the ev3dev kernel. So you would probably have to build your own kernel first to get that directory.

Sirus-7, 2025-01-31:
Because debian stretch is no longer in the current repository change in /etc/apt/sources.list

deb-src http://security.debian.org/ stretch/updates main contrib non-free
change to 
deb http://archive.debian.org/debian-security/ stretch/updates main contrib non-free

petusky, 2025-01-31:
Thanks @Sirus-7. However, I checked off that box earlier, per the instructions by @dlech, which instructed me to update  /etc/apt/sources.list to : 

```
deb http://archive.debian.org/debian stretch main contrib non-free
#deb-src http://archive.debian.org/debian stretch main contrib non-free

deb http://archive.ev3dev.org/debian stretch main
#deb-src http://archive.ev3dev.org/debian stretch main
```