pybricks/support #2040: [Bug] Hub crashes if exception is raised while handling unhandled exception.

dlech, 2025-02-01:
**Describe the bug**
The unhandled exception handler does not handle exceptions. We call `print_final_exception()` but the print functions can also raise an exception. Since there is no `nlr_push()` to handle any exception, we end up crashing the hub and it reboots (or powers off on hubs that can't keep the power up long enough for a reset).

https://github.com/pybricks/pybricks-micropython/blob/master/bricks/_common/micropython.c#L108

**To reproduce**

The easiest way to reproduce this is to modify `mp_hal_stdout_tx_strn()` to not actually print and and just cause an infinite loop calling `MICROPY_EVENT_POLL_HOOK`.

```c
        // pbio_error_t err = pbsys_bluetooth_tx((const uint8_t *)str, &size);
        pbio_error_t err = PBIO_ERROR_AGAIN;
```

Then download and run a program that prints something. (e.g. `print("hi")`)

The program will run forever because of the hacked print function. Press the stop button (hub button or Pybricks Code button) to stop the program. The firmware jumps to the function to print the unhandled `SystemExit` but gets stuck again in an infinite loop.

Press the stop button again. This time, there is nothing to catch the exception, so we end up in `nlr_jump_fail()`.

**Expected behavior**
This is how upstream MicroPython works, so maybe this is the expected behavior? But from a user point of view, having the hub go unresponsive and reset/power off isn't a great experience.

**Solutions considered**
1. We should make sure print functions can't get stuck in an infinite loop like this. (This was a problem while testing the USB support pull request).
2. We could disable the stop button from raising an exception for the duration of the unhandled exception handler.
3. We could add a `nlr_push()` to catch exceptions and silently ignore them during the unhandled exception handler.

Note: these are not mutually exclusive options.

laurensvalk, 2025-05-15:
4. Instead of passing `mp_plat_print.print_strn` to `mp_obj_print_exception(&mp_plat_print, exc)`, we could provide an equivalent function that calls `MICROPY_VM_HOOK_LOOP` instead of `MICROPY_EVENT_POLL_HOOK`, so can't raise another exception.

dlech, 2025-05-15:
I have a feeling that scheduled exceptions aren't the only exceptions that could possibly be raised in `mp_obj_print_exception()`. For example, if it allocates any memory, that can always raise an exception.

laurensvalk, 2025-05-17:
5. Different viewpoint: The majority of `mp_hal_stdout_tx_strn` implementations in MicroPython don't seem to call `MICROPY_EVENT_POLL_HOOK` at all. What are the downsides of calling just `MICROPY_VM_HOOK_LOOP` in ours?

Wouldn't scheduled exceptions still be raised after normal prints, just slightly delayed if it was a very long print? I think we've been over this at some point, but I don't quite remember why we settled on raising exceptions in the middle of a print.

dlech, 2025-05-17:
It would make `print()` uninterruptible and potentially block for a very long (CTRL+C wouldn't work). I don't think that would make for a good user experience when you print a giant string and regret it and want to interrupt the program. I know I've done that before a time or two. 😉

laurensvalk, 2025-05-17:
~~Lorem ipsum dolor sit amet, consectetur adipiscing elit. In porttitor aliquet ullamcorper. Duis sed nunc euismod, efficitur nulla ut, ullamcorper mi. Nam in sapien nisi. Vestibulum eu vestibulum enim. Pellentesque ipsum massa, lacinia eu fringilla ac, vehicula nec libero. Nunc molestie tempor arcu, in ullamcorper magna rutrum ut. Proin tristique tellus non nibh tristique maximus. Fusce pellentesque lorem eget elit lacinia, at ultricies sem faucibus.~~

~~Nam ut nulla tellus. Etiam et diam porta, bibendum leo vitae, efficitur ligula. Mauris fringilla felis eget turpis elementum, posuere suscipit magna pretium. Duis sagittis, libero sit amet suscipit scelerisque, felis elit euismod nibh, ut euismod sem massa nec neque. Integer vitae erat a augue tincidunt pretium sit amet eu sem. Mauris vitae aliquet risus. Integer egestas molestie enim sed scelerisque. Nulla placerat sapien vitae diam hendrerit, et maximus ante accumsan. Maecenas vel luctus mi. Cras sit amet erat vel massa auctor ultricies ut id arcu. Aliquam laoreet elit ac leo lacinia mollis eget in elit. Vivamus rutrum leo sit amet tempus eleifend. Aliquam non tincidunt ex, eu aliquet libero.~~

~~Suspendisse id sollicitudin enim. In mi justo, rhoncus eget diam nec, tempus viverra metus. Cras orci lorem, elementum viverra nisl quis, laoreet congue purus. Ut vel neque vel nisi congue rhoncus. Proin neque lectus, scelerisque eu nisl vel, commodo elementum nunc. Pellentesque condimentum nibh magna, et varius sapien cursus ac. Aliquam fermentum dolor id quam rhoncus placerat. Proin interdum eleifend sem in sollicitudin. Nam vitae mi ut ex pulvinar molestie. Lorem ipsum dolor sit amet, consectetur adipiscing elit.~~

~~Cras felis nisl, euismod vel volutpat a, tincidunt a lectus. Morbi faucibus lorem odio, ac suscipit nisi ultricies vitae. Nam porttitor consectetur ipsum, a tincidunt dolor euismod ut. Pellentesque maximus scelerisque ultricies. Praesent in vestibulum diam. Praesent tortor mi, tincidunt ut rutrum eu, rhoncus id mauris. Aliquam nec urna mollis, laoreet nibh et, mattis justo.~~

~~Etiam iaculis diam id ligula dictum, nec pretium diam faucibus. Nam non velit sed ligula laoreet dapibus ac at mi. In eget orci eu purus laoreet pretium. Nulla facilisi. Nam fringilla sem eget turpis sollicitudin sodales vel at magna. Aenean sagittis vitae magna sit amet facilisis. Vivamus tempor at elit nec ultricies. Duis faucibus finibus commodo. Morbi scelerisque diam quis faucibus venenatis.~~

~~Nunc vehicula eu purus eu aliquam. Ut egestas feugiat tellus, ut aliquet est tincidunt sed. Proin ut erat accumsan, mollis neque id, ullamcorper augue. Aliquam erat volutpat. Vivamus fermentum sapien nulla, a efficitur nibh placerat vel. Interdum et malesuada fames ac ante ipsum primis in faucibus. Nullam et porttitor metus, aliquet laoreet purus.~~

~~Fusce vitae pretium arcu. Nam felis diam, rutrum sit amet neque vitae, laoreet faucibus felis. Praesent sit amet nisl non risus rhoncus egestas. Cras ut ipsum vitae mauris faucibus luctus. In facilisis lacus ac velit volutpat, interdum pretium arcu luctus. Pellentesque nec neque metus. Nunc eget dolor id metus aliquet auctor. Lorem ipsum dolor sit amet, consectetur adipiscing elit. Phasellus laoreet tristique pretium. Morbi lacinia consectetur magna, quis aliquet urna. Suspendisse congue urna at fermentum facilisis. Etiam cursus maximus scelerisque.~~

~~Curabitur consequat consectetur turpis in placerat. Ut pretium nunc sit amet viverra tempor. Donec faucibus lectus eget erat maximus feugiat. Ut viverra ligula neque, sed egestas dolor dapibus vitae. Donec efficitur lacus vel viverra varius. Proin sollicitudin pharetra urna, eu convallis nulla euismod vitae. In interdum magna id sem porttitor, a vestibulum libero egestas. Suspendisse potenti. Praesent tincidunt venenatis felis, id dictum quam sagittis at. Etiam in dictum mi. Mauris gravida nibh at massa finibus, et vulputate orci interdum. Phasellus bibendum orci in orci vulputate rhoncus. Class aptent taciti sociosqu ad litora torquent per conubia nostra, per inceptos himenaeos. Proin vel leo est. Curabitur eget semper quam. Integer congue arcu sed nibh vulputate egestas ultrices eget justo.~~

~~Etiam porta at lorem in aliquam. Integer pretium porta libero, et eleifend lacus commodo at. Ut ac malesuada risus. Nullam sagittis fermentum lectus, ac efficitur turpis pellentesque vitae. Mauris a massa bibendum, blandit quam sit amet, pharetra ex. Mauris ornare rutrum risus, vel iaculis orci tincidunt at. Nulla facilisis libero sed purus mollis, at posuere odio faucibus. Fusce vitae nulla at justo lacinia varius a vel nulla.~~

~~Sed molestie tortor leo, quis cursus tellus commodo eget. Ut sit amet libero sapien. Praesent ac risus nec enim vulputate condimentum et quis turpis. Ut libero felis, mattis sed viverra id, accumsan eget ex. Morbi enim magna, lobortis in metus id, sollicitudin iaculis ipsum. Maecenas id commodo tellus, vel volutpat turpis. Vivamus metus nisi, mattis a quam sit amet, molestie tristique enim. Nunc eleifend, massa ut blandit vestibulum, mi augue laoreet neque, ut consequat ipsum odio in leo. Vestibulum risus tortor, condimentum lobortis auctor ac, pellentesque eget quam. Pellentesque posuere semper malesuada.~~

~~Pellentesque quis orci leo. Aenean ac mi est. Sed laoreet dolor eros, aliquam elementum tortor interdum sed. Nullam sagittis purus purus, id egestas purus efficitur blandit. Aenean cursus hendrerit lobortis. Phasellus malesuada, justo  [... cut ...]
~~Cras molestie augue a erat dignissim, ut tempus tellus maximus. Cras tristique odio erat. Etiam vestibulum metus vitae laoreet scelerisque. Praesent aliquet justo dolor. Phasellus nec purus eleifend, feugiat nisi quis, vulputate libero. Sed in fringilla ipsum. Integer gravida id mauris eget venenatis. Maecenas nec cursus quam. Nunc feugiat auctor libero, sit amet laoreet felis finibus in. Nullam vel dui id nisl lacinia pharetra eu dictum quam. Suspendisse eu orci eu leo fringilla hendrerit. Pellentesque vitae sollicitudin dolor. Proin dui metus, vehicula et rhoncus sit amet, condimentum vitae dolor.~~ I see what you mean.

laurensvalk, 2025-05-19:
Here is a good way to reproduce this on today's firmware without any hacks:

```python
def cursed():
    print("re", cursed())

cursed()
```

This prints a _very_ long stack trace. Pressing the stop button in the IDE or on the hub will crash it.

laurensvalk, 2025-05-19:
As part of the lesson on tasks, there was going to be a "don't try this at home!" side note. Rebooting the hub probably drives the point home but I would still rather fix it 😄 

[image]

laurensvalk, 2025-11-02:
> Here is a good way to reproduce this on today's firmware without any hacks:
> 
> def cursed():
>     print("re", cursed())
> 
> cursed()
> This prints a _very_ long stack trace. Pressing the stop button in the IDE or on the hub will crash it.

This no longer crashes now that David has added the `nlr` protection to the final exception print.

https://github.com/pybricks/pybricks-micropython/pull/409 ensures that the print comes through normally, and can be interrupted while adding a normal newline before and after:

[image]

So this can be closed.