pybricks/support #2045: [Bug] web app looses connection to hub, freezes, when trying to run trivial block program

m10d, 2025-02-04:
**Describe the bug**
Some time ago this worked on this tablet (android tablet, chromium-based default browser), just haven't used pybricks in a while.  Refreshed the page to ensure latest, about says `v3.5.0 (Pybricks Code v2.5.1)`

**To reproduce**
Steps to reproduce the behavior:
1. open code.pybricks.com
2. re-install pybricks fw just in case
3. start hub (blue light is `flash-flash-pause` for pairing)
4. connect bluetooth (blue light goes solid)
5. enter repl mode, blue light fadding in/out, `dir()` outputs reasonable stuff, can run `hello_world()` function
6. press stop button on web app.  Blue light goes back to solid.  
7. Repeat # 5 & 6 for good measure to ensure things seem to still be working after exiting repl --> same result
8. press run (`play` sym) on UI to run the program below
9. a second or so later, the hub light goes back to `flash-flash-pause` pairing.
10. app seems still responsive, but if I click around (say `settings` on left bar, then back to `files` on left bar) the app screen (well, web page within browser) goes blank white, and never comes back.  Browser itself is still functioning normally.
11. refresh the code.pybricks page; page loads, I can connect again to hub & start over; same result.

**Expected behavior**
runs the program

I tried a few other permutations of trivial block programs. 

**Screenshots**
[image]

dlech, 2025-02-04:
We had some report a similar problem (white screen in Pybricks Code) via email recently. I sounds like we might have an issue with the web server not serving the .wasm files that are needed to compile the code when doing download and run. This would explain why REPL works but download and run a program doesn't. We are looking into it.

m10d, 2025-02-04:
> We had some report a similar problem

If i missed something in my search of open issues, feel free to link & close this.

Thanks for your reply.

m10d, 2025-03-17:
Hoping for an update here - I've updated the (web) app and re-installed pybricks fw, and am still seeing the same issue ... which means I can't help my kid program

dlech, 2025-03-17:
We haven't had any reports of anyone else having problems for a while, so I thought the server issue was fixed. Maybe your problem is different after all.

Does the same problem happen on https://beta.pybricks.com?

m10d, 2025-03-18:
Yes, just tried beta.pybricks (including install fw via beta, jic it matters). Same result, when "run" hub goes from solid blue to flashing blue, ui eventually becomes whitescreen.

-------- Original Message --------

dlech, 2025-03-18:
The fact that the hub light changes tells me that either the hub is crashing and rebooting or that the host computer Bluetooth isn't happy and is disconnecting. 

If you could capture Bluetooth packets and share that with us, we could see if that is the problem. You can find some instructions on how to do that at https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture-bluetooth-traffic

dlech, 2025-03-19:
And for the white screen issue, if you could open the Developer Tools in your browser and check the console and network tabs (for network, you will need to do a page refresh) to check for errors. If you need more help figuring out how to do that, let me know.

Or if you have a different computer you can try, that could be helpful too.