pybricks/support #2048: [Bug] I can broadcast list via BLE, but not 2D list.

romw314, 2025-02-05:
**Describe the bug**
When I try to broadcast a 2D list (not too big), it says:
```
TypeError: must be True, False, int, float, str or bytes
```

**To reproduce**
Steps to reproduce the behavior:
1. Create a new file in [Pybricks Code](https://code.pybricks.com).
1. Paste the following code:
```python
from pybricks import version
from pybricks.hubs import ThisHub

print('pybricks:', version)

ThisHub(broadcast_channel=0).ble.broadcast([0,1,[2,3]])
print('broadcasted 2d list')
```
3. Run the program on a hub.
1. You will get output like the following:
```
pybricks: ('cityhub', '3.5.0', 'v1.20.0-23-g6c633a8dd on 2024-04-11')
Traceback (most recent call last):
  File "BLE_arr.py", line 6, in <module>
TypeError: must be True, False, int, float, str or bytes
```

**Expected behavior**
It should output:
```
pybricks: ('cityhub', '3.5.0', 'v1.20.0-23-g6c633a8dd on 2024-04-11')
broadcasted 2d list
```

**Screenshots**
[image]

laurensvalk, 2025-02-05:
> TypeError: must be True, False, int, float, str or bytes

As the message indicates, the values in the list must be True, False, int, float, str or bytes.

So, you can't have a list in a list when broadcasting. So instead of broadcasting `[0, 1, [2, 3]]`, you can broadcast `[0, 1, 2, 3]`, and design your code to work that way.

It is also best to avoid this:

```python
ThisHub(broadcast_channel=0).ble.broadcast([0,1,[2,3]])
```

And instead do:

```python
hub = ThisHub(broadcast_channel=0)
hub.ble.broadcast([0,1,[2,3]])
```

romw314, 2025-02-06:
> As the message indicates, the values in the list must be True, False, int, float, str or bytes.
>
> So, you can't have a list in a list when broadcasting. So instead of broadcasting `[0, 1, [2, 3]]`, [...]

Thank you for your reply. I understand the error now, but it says that must be True, False, int, float, str or bytes and doesn't clarify what must be True, False, int, float, str or bytes, so I thought that the value I'm sending via BLE must be True, False, int, float, str or bytes. I can always use JSON, which, on the other side, includes markup and I can send only 24 bytes of data.

I also wrote a library (https://gitlab.com/romw314/pybricks-lib/-/blob/master/rolocomm.py?ref_type=heads) which sends always bytes (JSON), and sends data multiple "packets", so the data isn't limited to 24 bytes (errors show 26, but I figured out that there is a 2-byte head). This library could be better, but it uses JSON too, so it supports 2D lists.

> It is also best to avoid this:
> 
> ThisHub(broadcast_channel=0).ble.broadcast([0,1,[2,3]])
> And instead do:
> 
> hub = ThisHub(broadcast_channel=0)
> hub.ble.broadcast([0,1,[2,3]])

I just did this because I want to have the example as simple as possible.

In a real program, I assign it to a variable even if I use it only once.

closed 2025-02-06 by romw314