pybricks/support #2068: [Bug] Run button greyed out

poopmaster400, 2025-02-17:
The Run button is greyed out when a hub is connected with the new firmware and wont let me run any code at all.

On reproducing the problem i have no clue, i still cant even get it back to normal

[image]

The code visible is the code i ran before it broke, i dont know or think it is the problem

laurensvalk, 2025-02-17:
Does this persist if you close the browser, and try again?

poopmaster400, 2025-02-17:
> Does this persist if you close the browser, and try again?

Yes it does, the same issue

laurensvalk, 2025-02-17:
Was it working before? Does it make a difference if you try it with https://beta.pybricks.com/, installing the firmware from there as well?

poopmaster400, 2025-02-17:
> Was it working before? Does it make a difference if you try it with https://beta.pybricks.com/, installing the firmware from there as well?

It was working before and i just tested with a linux device. Pairing with the same hub works just fine.

Running on beta firmware results in the same issue. 
To mention i am doing this on a school controlled Chromebook, but since it was working fine i dont know if that is the issue

dlech, 2025-02-17:
Is it possible that there is a program running on the hub?

The run button state depends on two things. A) that there is an active file open and B) that the hub state is "idle". The "idle" state depends on if we think some command that starts or stops a program is in progress or not in the IDE or if the hub tells us a user program is running.

I have a Chromebook I could try later to see if that has something to do with it.

laurensvalk, 2025-02-17:
To be clear, does this happen right after connecting, or right after clicking the play button to start it?

> Is it possible that there is a program running on the hub?

The stop button should be enabled in that case.

What could be happening is that the hub is not sending status messages or Pybricks Code is not receiving them. I've also seen this sometimes after starting a program, but usually not consistently.

poopmaster400, 2025-02-17:
> To be clear, does this happen right after connecting, or right after clicking the play button to start it?
>
>
> The stop button should be enabled in that case.
>
> What could be happening is that the hub [...]

Sorry for late reply. Yes it happens right after connecting and i have reinstalled the official firmware and the pybricks firmware many times so that would rule an active program out. right?

I have got no clue as to why this is happening. Well i need to sleep, i can maybe reply tomorrow sometime

dlech, 2025-02-17:
> What could be happening is that the hub is not sending status messages or Pybricks Code is not receiving them.

I think this is the mostly likely reason. It could be that something changed in the ChromeOS Bluetooth stack that causes BLE notifications to not work.

poopmaster400, 2025-02-18:
> > What could be happening is that the hub is not sending status messages or Pybricks Code is not receiving them.
> 
> I think this is the mostly likely reason. It could be that something changed in the ChromeOS Bluetooth stack that causes BLE notifications to not work.

Maybe, and fixing that would be hard i assume

laurensvalk, 2025-02-25:
Possibly related, but hard to say for sure.

[image]

poopmaster400, 2025-03-06:
> Possibly related, but hard to say for sure.
> 
> [image]

Well, for some reason it works now. I think it was only a temporary issue.

closed 2025-03-06 by poopmaster400