pybricks/support #2081: [Feature] Backwards compatible block updates

laurensvalk, 2025-02-21:
**Is your feature request related to a problem? Please describe.**
We will be introducing a few new blocks and adapt a few existing ones. Like making curves in https://github.com/pybricks/support/issues/1157.

With other Blockly-based apps like the official SPIKE app, this often results in broken user code because of broken compatibility. "Heads up, could not open program".

I would like to avoid breaking user space at all cost 🙂.

**Describe the solution you'd like**
While it is tempting to invent some automatic update procedure, it seems to me that this is asking for trouble.

If we really have to change a block, I think we can roll out a completely new block. The old block continues to exist and generate valid code, but it won't be visible on the block palette anymore.

We could add a notice on the old version of the block:

[image]

On the other hand, it doesn't seem very user friendly.

Perhaps we should only show it if the user is using the feature that is being phased out, like the curve method.

dlech, 2025-02-21:
I agree this is the most sensible course of action. Bonus points if the warning icon points to documentation on how to replace the deprecated block with the "new" block or automatically generates the replacement with the appropriate parameters filled in (if that is even possible).

erik-thompson-seattle, 2025-02-21:
As I have not formally reviewed the codebase/architecture around Pybricks and rules, this will change in the coming months as I have acquired ~30 surplus Ev3 units, I may be proposing something that doesn't address the concern. 
For consideration, could you create a Major release update of Pybricks introducing variable args support? Then, any unsupported args/values are sent to debug output, and the program keeps working with what is supported rather than "Heads up, could not open program", which I assume happens on the device.

laurensvalk, 2025-02-26:
@erik-thompson-seattle -- I'm not sure exactly what you mean. The new Pybricks version with blocks is not yet available for EV3. Feel free to open a dedicated discussion with your questions.

laurensvalk, 2025-02-26:
The feature discussed here is available in Pybricks Beta now.

closed 2025-02-26 by laurensvalk

erik-thompson-seattle, 2025-02-28:
> [@erik-thompson-seattle](https://github.com/erik-thompson-seattle) -- I'm not sure exactly what you mean. The new Pybricks version with blocks is not yet available for EV3. Feel free to open a dedicated discussion with your questions.

Thanks @laurensvalk. I had not looked closely at the supported hubs, but I thought I would be able to use it with the EV3 units and contribute to the Pybricks repositories. I will still contribute but need to get the Lego E3 firmware first to get my infrastructure and classroom material up and running rather than being literal bricks.

Regardless of EV3, I was suggesting a major vs minor release of pybricks where the APIs under the blocks supported variable args and to assist with legacy support and future support in which it could report that options were not supported when the program ran rather than not run at all.